greatscottgadgets/hackrf #379: what's  accuracy of  SDR oscillator in HackRF one? for example, the USRP1 to 20 parts per million (ppm)

stackprogramer, 2017-05-18:
i read in  this link that USRP1 has accuracy frequency clock??

>  for example, the USRP1 to 20 parts per million (ppm)his is sufficient for a number of purposes, but not for others. Some SDRs (including the HackRF and USRP) also support external oscillators. For example, the Clock Tamer, the KSP 52MHz and the FA-SY 1 are all accurate clocks.

https://github.com/scateu/kalibrate-hackrf

i want to know whats this for HackRF one ? it can effect on bad receiving HackRF one?
thanks best regards stackprogramer

dominicgs, 2017-05-24:
HackRF One features a crystal that has frequency stability of 20 ppm.

closed 2017-05-25 by dominicgs

schielb, 2022-04-28:
@dominicgs Where did you find that out?