greatscottgadgets/hackrf #380: Problem with HackRF after flashing new firmware

korczis, 2017-05-20:
When I received HackRF everything worked as expected. When I flashed portapack firmware and later latest HackRF firmware it started to misbehave.

When I connect the HackRF to my mac it is not recognized. I need to run dfu-util and flash the firmware first. After reset it is not recognized and I need to flash the firmware again.

### Steps to reproduce
1. Connect HackRF to usb 

### Expected behaviour

HackRF should be recognized.

### Actual behaviour
HackRF is not recognized and I need to boot it in DFU mode and flash the firmware.

### Version information
**Operating system**:
macOS Sierra
version 10.12.4

**Hardware**
MacBook Pro (13-inch, 2016, Four Thunderbolt 3 Ports)

**hackrf_info output:**

After reset (see attachment `leds_after_reset.png`):

```
$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
No HackRF boards found.
```

**Workaround**

1. Boot HackRF in DFU mode
2. Flash latest firmware
3. Run hackrf_info

*Flash firmware (see attachment `leds_after_flash`)*

```
$ dfu-util --device 1fc9:000c --download hackrf-2017.02.1/firmware-bin/hackrf_one_usb.dfu  --reset
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        21992 bytes
Download done.
dfu-util: unable to read DFU status after completion
dfu-util: can't detach
Resetting USB to switch back to runtime mode
```

*Run hackrf_info after flash*

```
$ hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c8226fac5f
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00654f69
```
### Attachments

**LEDs after reset**

[image]

**LEDs after firmware flash/dfu**

[image]

**Recognized peripherals after reset**

[image]

[usb-after-reset.txt](https://github.com/mossmann/hackrf/files/1016318/usb-after-reset.txt)

**Recognized peripherals after firmware flash/dfu**

[image]

[usb-after-flash.txt](https://github.com/mossmann/hackrf/files/1016319/usb-after-flash.txt)

**Box**
[image]

jboone, 2017-05-20:
Tomas,

This is more than likely my fault. I have been slow producing an official release of the PortaPack firmware that incorporates the new HackRF firmware. The new HackRF host tools apparently require a matching firmware version. I have a test binary that I prepared a couple of months ago, but have had little feedback on the outcome. Please try it and let me know if it addresses this issue. The URLs to download the test build are in the last comment of this GitHub issue:

https://github.com/sharebrained/portapack-hackrf/issues/108

        - Jared

korczis, 2017-05-20:
@jboone, I tried this but HackRF is still not recognized after reset :(

dominicgs, 2017-05-22:
After successfully running the DFU tool and hackrf_info, could you try running `hackrf_spiflash -w hackrf_usb.bin` ?

theforcedk, 2017-06-02:
I had this issue also, on Sierra. I had to re-flash an older (non github master) version of the hackrf_usb.bin to get it working again.

hagensieker, 2017-07-01:
I am having the exact same issue.  After trying to update firmware on hackrf one I bricked it.  I can DFU to install, hackrf_spiflash to write firmware and it will not persist after a reboot.  How can I write firmware that will work after unplugging or a reset?

korczis, 2017-07-01:
@hagensieker in my case there was HW problem so I had to sent HackRF for replacement.

hagensieker, 2017-07-01:
Thank you!  I have had mine about 2 days and got it from Amazon so I just now sent it back and ordered a new one.

If you received your new one did it accept the firmware okay or did it come with the latest firmware?

korczis, 2017-07-01:
@hagensieker it has not arrived yet.

hagensieker, 2017-07-01:
@korczis I'm set to get my replacement on Monday.  I'm a bit gun shy to try another firmware upgrade if mine shows with 2015 07 firmware.

dominicgs, 2017-07-03:
I'd highly recommend updating to the latest firmware as the majority of firmware updates on HackRFs are successful.  In a very small number of cases the flash chip (which stores the firmware) on HackRF fails, in those cases we will replace the device through the original reseller.

dominicgs, 2017-07-03:
I'm closing this issue as you both have replacements on their way.

closed 2017-07-03 by dominicgs

hagensieker, 2017-07-05:
Just received my replacement. Came with 2015 Firmware.  New 2017 02.1  Firmware upgraded okay and cpld also.  Works perfect.

dominicgs, 2017-07-06:
That's great to hear. I'm sorry that your first unit failed the firmware
upgrade.

master7xx, 2017-07-06:
And my awesome story:
got one board from nooelec with firmware 2015 and two board from china with firmware 2017 2 1.
Upgrade on board from nooelec failed with symptoms like above.

a5ync, 2017-07-13:
> hackrf_info version: unknown
> libhackrf version: unknown (0.5)

It is very confusing for the new owners (like me) of HackRF. It actually has nothing to do with the device itself and kinda forced me to get two of these units... I got one from an unknown source and was not sure if it is broken or not. Turned out to not to be, but in the meantime I got another one from NooElec that was giving me the same exact output.. Anyway, both units work great under Ubuntu 16.04.02 LTS
The "unknown" output of hackrf_info is only showing up on my OSX installation since the linux version doesn't include these two lines of the output at all. All the firmware updating procedures worked fine and devices were detectable by OS. Also I am curious - those of you that received new units - did it change the output of hackrf_info to be different than "unknown" ?

korczis, 2017-07-13:
@a5ync I can not tell because I flashed replaced unit with different firmware - fork of this repo.

https://github.com/furrtek/portapack-havoc

dominicgs, 2017-07-13:
@a5ync I closed the issue because I thought it was confusing, and I was hoping people wouldn't see it as easily!

The 'unknown' lines are for hackrf_info and libhackrf version strings.  The OS X build process (I'm assuming homebrew or ports?) doesn't appear to set these strings so they come up as unknown.  I'll need to look in to that before the next release.

a5ync, 2017-07-14:
@dominicgs thank you for the validation. I figured it is not a hardware issue so wanted to leave a note for others before they are going to return their devices as broken. Keep up the good work!

Marcka, 2017-09-03:
I'm having similar issues, updated to this new firmware and the device wouldn't boot. Ran the DFU and the device throws hackrf_info and libhackrf unknown lines but will successfully flash the older firmware from 2015. Otherwise whenever I reboot the device is undetected until I rerun DFU and update from ram. The rom wont hold on reboot of this version but will for the 2015.

Marcka, 2017-09-03:
@dominicgs getting the same errors but on Kali, anything worth testing?

dominicgs, 2017-09-06:
@Marcka I'm working on a test tool to check the status of the flash chip.  Would it be possible for me to send it to you so that you can check your HackRF's flash chip?

Marcka, 2017-09-08:
@dominicgs Happily, let me know.

dominicgs, 2017-09-13:
@Marcka I'm added the ability to read the flash chip status register to my branch here:https://github.com/dominicgs/hackrf/tree/flash-investigation

It requires building firmware and host code for your system.

ihorman, 2022-09-11:
@dominicgs  could you please share with me a tool to check the flash rom on hackrf, I have exactly the same problem

miek, 2022-09-11:
> could you please share with me a tool to check the flash rom on hackrf? I have exactly the same problem as written above. Thanks in advance!

Please could you open a new issue with all the details of the problem you're seeing, and we'll work through it there. As this issue is 5 years old and we've had 2 releases since, trying to work from advice here is likely to confuse things.