greatscottgadgets/hackrf #383: Two HackRF One Doesn't work in Gnuradio Simultaneously

stackprogramer, 2017-06-08:
In gnuradio 3.7.12 we have two hackrf one, we want to see spectrum we test in Debain jessie and Ubuntu linux all of them can not play two HackRFOne   Simultaneously.
you can see video on youtube for more info....
 https://www.youtube.com/watch?v=xu9qPR_QH0E
can we receive data from two hackrfone with osmocomsource   Simultaneously
please help
best regards stackprogramer

dominicgs, 2017-06-08:
It looks like both HackRFs produce data, but not simultaneously.  Is that correct?

At first glance this appears to be a GNU Radio scheduler issue, but I'm happy to look in to it.  Could you supply any output from running the flowgraph?  Are there error messages?  Do you get Overrun warnings?

Additionally, I recommend specifying the HackRF using the serial number rather than the index.

stackprogramer, 2017-06-09:
@dominicgs  Hi, this is not correct HackRFs produce data, but not simultaneously. Is that correct?
when i run gnuradio randomly on of them works , other one doesn't work...........
i rerun gnuradio in video youtube. it has not any strange error message,i created a issue too on gnuradio https://github.com/gnuradio/gnuradio/issues/1326 .
i will test serial number soon and share result
thank you

dominicgs, 2017-06-09:
Is it always the same one that works?

stackprogramer, 2017-06-10:
@dominicgs no, randomly one of them will work.
gnuradio closed my issue and talk that is related to osmosource, i will soon examine more and share new result too. thank you

stackprogramer, 2017-06-15:
Hi,my problem finally is solved,i  was setted sps to 32k, by changing this to 1M sps now it works like charm,in low sps it has problem, so my problem is solved 
thanks very much

closed 2017-06-19 by dominicgs