greatscottgadgets/hackrf #384: hackrf_info fails with undocumented error message on Arch Linux

yschroeder, 2017-06-12:
### Steps to reproduce
1. install software (packages `gnuradio`, `gnuradio-osmosdr`, `gnuradio-companion`)
2. run `hackrf_info`

### Expected behaviour
`hackrf_info` shows information about the connected HackRF One.

### Actual behaviour
`hackrf_info` returns obscure error message that is not documented anywhere (even google finds nothing)

### Version information
**Operating system**:
`Linux *HOSTNAME* 4.11.3-1-ARCH #1 SMP PREEMPT Sun May 28 10:40:17 CEST 2017 x86_64 GNU/Linux`

**hackrf_info output:**
```
# hackrf_info
hackrf_info version: 2017.02.1
libhackrf version: 2017.02.1 (0.5)
Found HackRF
Index: 0
hackrf_open() failed: Input/Output Error (-1000)
```

### How to resolve
```
# modprobe hackrf
```
The hackrf kernel module was not loaded (and the documentation for Arch Linux did not tell to do so).
Afterwards, `hackrf_info` works as expected

```
# hackrf_info
hackrf_info version: 2017.02.1
libhackrf version: 2017.02.1 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c83358b0cf
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: 0xa000cb3c 0x00654753
```

The wiki page on operating system tricks should maybe point to this command at least for Arch Linux.

dominicgs, 2017-06-12:
This is an interesting solution but I'm concerned that it's masking a different problem.

The hackrf kernel module is not part of this project and is not required for HackRF to function.  However, it is possibly solving a power management or similar issue.

I'd like to keep this issue open until I can track down what may be causing this.

yschroeder, 2017-06-12:
Power issues seem plausible. I am observing that `hackrf_info` does not work every time. It still fails once and then works multiple times in a row.

Executing the flow graph in GRC works strictly every second time. It fails once (without crashing or error message, just not returning no data at all from the osmocom source), then works the next time, then fails again...

dominicgs, 2017-06-12:
Are you running any power management tools, such as TLP or laptop-modes?

yschroeder, 2017-06-12:
Yes, it actually was TLP and blacklisting as described in the wiki FAQ worked for me. However, for people not familiar with configuring TLP (read: me) that FAQ is a bit short. It should point to the correct file and tell how to restart TLP. The config file is `/etc/default/tlp` and restarting is done via `sudo systemctl restart tlp.service` (at least for me on Arch linux).

Also it could mention the exact error message I reported in the first post.

yschroeder, 2017-06-12:
I disabled that kernel module again and it still works. It was the USB autosuspend issue.

dominicgs, 2017-06-12:
That's great!  I agree that the FAQ is a little lacking here, I'll take a look at updating it today

dominicgs, 2017-06-12:
I've update the FAQ here: https://github.com/mossmann/hackrf/wiki/FAQ#power-saving-and-usb-autosuspend

Let me know if you think it needs further edits or clarification.

closed 2017-06-12 by dominicgs

yschroeder, 2017-06-12:
I think the tlp binary has no parameter restart, it has only start and stop. But I do not know for sure, I'll check tomorrow. Otherwise write "restart tlp service" and people will be able to do that via whatever they use (e.g. systemctl). Otherwise your updated FAQ is fine.