greatscottgadgets/hackrf #385: hackrf tools built with VS

jc1122, 2017-06-17:
Hello,

I built the new release (2017.2) of hackrflib and tools using VS2017 and copied the built DLL's as well as the tools to one folder. I ran hackrf_info on 2  PC's with Windows 7. On one of them, everything is fine. On the other I get the 0xc00007b "Application was unable to start correctly" error.  Most probably there is a mismatch in DLL files in the system (32 bit instead of 64 bit). 
Which VC libraries are hackrf tools using?

Hackrf tools built with Cygwin work correctly on both machines.

dominicgs, 2017-06-22:
I'm not familiar with Windows development or the 0xc000007b error, but some searching suggests that it a dependency is missing.  There are a couple of dependencies for libhackrf, pthreads and libusb.  I would start by checking that both DLLs are available.

You could use [Dependency Walker](http://www.dependencywalker.com) to find the missing dependencies.

jc1122, 2017-06-23:
Thank you!  Dependency Walker indicated, that MSVCR110.dll and MSVCR100.dll were 32 bit instead of 64 bit. After replacing these files to 64 bit, everything works fine.

closed 2017-06-23 by jc1122