greatscottgadgets/hackrf #389: TX power

marcoD64, 2017-07-05:
I use HackRF One for transmitting a gps simulation (https://github.com/osqzss/gps-sdr-sim) and I like to know the power in dBm of the TX port. So what is it or how can I measure it?

stackprogramer, 2017-07-06:
hi, you can refer to wiki page hackrfone https://github.com/mossmann/hackrf/wiki/HackRF-One 
there is mention these info, for practical you should use spectrum analyzer or network analyzer.

you can use two SDR hardware instead using port, using  two antenna and see spectrum on gqrx sotfware.
best  regard stackprogramer

orrinwinton, 2018-06-13:
I'm thinking of getting one of these meters, and hook one end to the HackRF, the other end to a 50-ohm dummy load. Any comments on this from the experts, please:

https://www.ebay.com/itm/1-500Mhz-OLED-RF-Power-Meter-75-15dBm-Power-Set-RF-Attenuation-Value-battery/132248196401?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2055119.m1438.l2649

closed 2019-11-25 by miek