greatscottgadgets/hackrf #390: observing Random Jitter in HackRFone in High Bandwidth with high carrier modulation

stackprogramer, 2017-07-06:
Yet i work on DVB-T in HackRF one, in TX we can send DVB-T and receiver with a hardware dongle DVB-T. but for receiving with SDR Based we had error:
http://blog.stackprogramer.xyz/en/2017/02/18/dvbt-project-receiver-probelm-oorestart-aquisition-d_freq_offset-5-od_freq_offset-5/

> DVBT project receiver probelm OOrestart aquisition d_freq_offset: -5 Od_freq_offset: -5

error happening it is increased with increasing sps.....

so we can see in dvb-t block in gnuradio we have a block that is named:dvbt_ofdm_sym_acquisition

https://gnuradio.org/doc/doxygen/classgr_1_1dtv_1_1dvbt__ofdm__sym__acquisition.html#details
so we  can see source cpp of this block you can see it in this link:
https://github.com/gnuradio/gnuradio/blob/master/gr-dtv/lib/dvbt/dvbt_ofdm_sym_acquisition_impl.cc

this source want to estimate offset frequency but we can see this offset is random and change instantly it is not fixed it is random...........
http://www.eetimes.com/document.asp?doc_id=1204338

i think we face with a random error that is name random jitter, it is relation to thermal noise , in higher rate or sps we increase effect thermal noise , it is fuck up error ......
random jitter is overcome in hackrfone
https://www.maximintegrated.com/en/app-notes/index.mvp/id/2875

stackprogramer, 2017-07-06:
see more info, in my blog post http://blog.stackprogramer.xyz/en/2017/07/06/observing-random-jitter-in-hackrf-one-in-high-bandwidth-with-high-carrier-modulation/

mossmann, 2017-07-12:
As was pointed out on Discuss-gnuradio, you haven't provided any evidence that supports your conclusion that there is a jitter problem.  I suggest analysis of gr-dvbt to determine the cause of the error.

closed 2017-07-12 by mossmann

stackprogramer, 2017-07-13:
@mossmann  Hi, we with gr-dvbt can receive and transmit with USRP, i think we should examine HackRF One's Parts for detecting problem.we after some examine we concluded that it related to random jitter error, if we find new evidence we will share. thanks

stackprogramer, 2017-08-10:
in high bandwidth we can not receive dvb-t and in lower bandwidth 1MB we have bad quality video.

https://www.youtube.com/watch?v=HhOoBXZPjk4
http://blog.stackprogramer.xyz/en/2017/08/10/dvb-t-gnuradio-tx-rx-recording-video-receiver-in-bandwidth-1mb-and-frequency-461mb/