greatscottgadgets/hackrf #392: Receive error using GNU Radio

TheCrippledModder, 2017-07-09:
### Steps to reproduce
1. Make a flowgraph in GNU radio with a FFT sink
2. Execute flow graph
3. Kill flow graph using kill flowgraph button

### Expected behaviour
RX light should stop being enabled on the HackRF

### Actual behaviour
RX light stays ON until reset or power off

Side note: if I exit the FFT flowgraph using it's GUI exit button "X" the RX light turns off

### Version information
**Operating system**:
Pentoo 4.4.8

++ Not sure if it is an error with GNU Radio or firmware ++
++ Did not try using other firmware updates yet ++

dominicgs, 2017-07-11:
This is relatively common, there is a bug somewhere that means that we don't cleanly shut down the HackRF when using the kill flowgraph button.  I haven't been able to track it down so far.

I recommend using the GUI's 'X' button as a workaround.

miek, 2019-12-09:
Duplicate of #180

closed 2019-12-09 by miek