greatscottgadgets/hackrf #396: Firmware Recovery Issue

jringley-mhia, 2017-07-21:
Thanks in advance for any support!

### Steps to reproduce
(Newest firmware release.) 
1. Not sure what happened but apparently had issue compiling firmware for HackRF One. My HackRF One was working but had old firmware. Compiled firmware and flashed the file to the device, now when powered on HackRF cannot be found by utilities. 

The HackRF only has two LEDs lit (steady): 3V3 and TX

Since the initial issue with the firmware I compiled I have started using the newest firmware release (2017.02.1) to rule out any potential compilation issues.

2. Booting into DFU mode allows me to install the DFU firmware to RAM. hackrf_info and other utilities now work correctly. Now the HackRF LEDs are as follows (steady) 3V3, 1v8, RF, USB.

3. I can now use hackrf_spiflash utility to install the firmware to flash. No Error messages. No change in LED state. hackrf_info and other utilities continue to work just like they did after installing DFU firmware.

After both steps 2 & 3 I can begin using the HackRF One without issue (to the best of my knowledge), until I reset or power off the device. At that point it returns to the state of #1 above: no hackrf utilities can find it, only 3V3 and TX LEDs are lit.

Please note that I have also installed the CPLD (procedure: first installing DFU, then SPI, then CPLD.) No error messages, the 3 lights begin blinking on CPLD installation, the terminal says write was successful and prompts me to reset or unplug and plug device back in.... Same as above, as soon as reset or device loses power, device is back to no firmware.

### Expected behaviour
According to the firmware recovery instructions I think I should have fixed any firmware issues after installing SPI and the device should work normally after being powered off.

### Actual behaviour
As stated above pressing reset or powering off returns the HackRF One to a non-working state.

It appears as though the RAM isn't being saved to the ROM so a reset or loss of power clears the firmware from the RAM?

### Version information
**Operating system**: pentoo-amd64-hardened-2015.0_RC5 (LiveCD)

**hackrf_info output:**
(Output after flashing DFU or DFU & SPI):

pentoo@pentoo ~/Desktop/hackrf-2017.02.1/firmware-bin $ hackrf_info
Found HackRF board 0:
USB descriptor string: 0000000000000000a06063c82448915f
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1
Part ID Number: 0xa000cb3c 0x00714f6a
Serial Number: 0x00000000 0x00000000 0xa06063c8 0x2448915f

(Output after resetting or unplugging and plugging back in):

No HackRF boards found.

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

N/A

### Output
```
Insert any commandline or build output here
```

jringley-mhia, 2017-07-21:
Just wanted to add that I am following the instructions from this page:

https://github.com/mossmann/hackrf/wiki/Updating-Firmware

Thanks again!

dominicgs, 2017-07-21:
It sounds like an issue writing to the flash chip.  Could you try `hackrf_spiflash -r firmware.bin` to read out the existing firmware?

jringley-mhia, 2017-07-21:
dominicgs - Thanks for the reply! Should I do this after the DFU boot and DFU firmware installation? Or after the DFU installation then the SPI installation?

Either way I'll give it a shot when I get home from work later.

dominicgs, 2017-07-21:
Good question, I should have been specific.

Could you do both before and after running 'hackrf_spiflash'?  Then compare the files?

backahasten, 2017-08-23:
I got the same.maybe the season is i write a wrong file.What should i do when the dfu is useless.

sincars, 2017-09-15:
Hello. I have exactly the same issue. I also did as Dominicgs suggested and In my opinion both files look same.  But im not specialist to verify.
I have also second HackRF One unit and I had no issues upgrading firmware. I also used this page for upgrade instructions: https://github.com/mossmann/hackrf/wiki/Updating-Firmware

Thank You.

dominicgs, 2017-09-15:
@sincars Are you able to build a new DFU firmware and `hackrf_spiflash` tool from my [branch](https://github.com/dominicgs/hackrf/tree/flash-investigation)?

I'd like to read the status registers from your flash chip, but I've only just added the feature.

sincars, 2017-09-21:
Hello dominicgs. There are too many things to install and too many dependences and I get too many errors on every step. I just dont have time and knowledge to solve them. Unless there is available already compiled DFU firmware. Anyway, looks like warranty case for me and I just send the unit back.

dominicgs, 2017-09-21:
@sincars I completely understand.  Go ahead and contact the reseller that you bought the HackRF from.

I'll try to get this change merged in to the main HackRF repository to help people with this issue in the future.

closed 2017-11-16 by dominicgs

digrx, 2019-04-04:
> Good question, I should have been specific.
> 
> Could you do both before and after running 'hackrf_spiflash'? Then compare the files?

I have same problem and I did this.
both files are same, but after reset it still no firmware/non-working state

4JAM4, 2021-02-21:
HI,
I have the same issue, and am trying to recover the SPI Flash Firmware in DFU mode. 

There seems to be a problem with "hackrf_one_usb.dfu"  ===>  "No such file or directory"

Please refer below:
"
andrew@pi-sdr:~/Desktop$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: Could not open file hackrf_one_usb.dfu for reading: No such file or directory
andrew@pi-sdr:~/Desktop$ 
"
Please note that I am following the directions at:
https://mirrors.spacecruft.org/mossmann/hackrf_wiki/src/branch/master/Updating-Firmware.md