greatscottgadgets/hackrf #398: Losing hackrf usb detection after bad firmware update

darcosion, 2017-08-06:
### Steps to reproduce
With a hackrf one, just : 
1. sudo hackrf_spiflash -w hackrf_jawbreaker_usb.bin

### Expected behaviour
Nothing, it's an accident with TAB

### Actual behaviour
$ hackrf_info 
hackrf_info version: git-9bbbbbf
libhackrf version: git-9bbbbbf (0.5)
No HackRF boards found.

### Version information
**Operating system**: Linux 4.4.0-89-generic Ubuntu

dominicgs, 2017-08-08:
It looks like you have flashed the Jawbreaker firmware to your HackRF One.  Unfortunately the two boards are not able to share the same firmware.

There is a recovery mode, which can be read about here: https://github.com/mossmann/hackrf/wiki/Updating-Firmware#only-if-necessary-dfu-boot

It boils down to holding the DFU button while connecting your HackRF to the USB cable and then running:
```
sudo apt-get install dfu-util
dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
hackrf_spiflash -w hackrf_one_usb.bin
```

darcosion, 2017-08-08:
I have tried this solution and the second commande doesn't work : 
`/hackrf-2017.02.1/firmware-bin$ sudo dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available`

darcosion, 2017-08-08:
It's my cable !

I have changed my cable and the command DFU work.
It's really strange the fact than some cable work totaly and other not, I think it's related to the norme of USB-C.

closed 2017-08-08 by darcosion

dominicgs, 2017-08-09:
For what it's worth, I wrote a firmware compatibility check in to hackrf_spiflash this morning: https://github.com/dominicgs/hackrf/commit/75d3a0a5b1ccdf9064fab5f369f4413c623b526f

This should prevent this error in the future, although I'm not super confident about the robustness of the check.