greatscottgadgets/hackrf #401: hanging on close()

jpetrescu, 2017-08-14:
### Steps to reproduce
1.run hackrf_transfer
2.ctl c
3.

### Expected behaviour
hackrf_transfer should end normally, and it does quite often. USB comm ends
 before complete ending of hackrf_transfer.

### Actual behaviour
stop() successful
usb dropped (LED goes off)
stop() sent
hangs
sometimes comes back with -1000 error, then hangs
--output--
C:\hackrf\host\cmake\hackrf-tools\src\Release>hackrf_transfer -l 8 -g 20 -s 20000000 -f 125000000 -r test.iq
lna_gain=8
vga_gain=20
sample_rate_hz=20000000
freq_hz=125000000
path=test.iq
###baseband_filter_bw_hz=0
Found 1 hackrf board(s).
call hackrf_set_sample_rate(20000000 Hz/20.000 MHz)
call hackrf_set_freq(125000000 Hz/125.000 MHz)
Stop with Ctrl-C
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.001 sec = 40.1 MiB/second
Caught signal 0
40.1 MiB / 1.001 sec = 40.1 MiB/second

Exiting...
Total time: 3.00440 s
hackrf_stop_rx() done
------USB LED is off--------when I press the reset button about half the time I get
hackrf_close() failed: Input/Output Error (-1000)
hackrf_exit() done
fclose(fd) done
exit

### Version information
**Operating system**: Windows 10

**hackrf_info output:**
C:\hackrf\host\cmake\hackrf-tools\src\Release>hackrf_info
hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c82344745f
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: 0xa000cb3c 0x00664f5e
REACHED THE END

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output

Insert any commandline or build output here
```hackrf & hackrf-tools for Windows built from instructions found on-line

dominicgs, 2017-11-09:
Sorry, I seem to have missed this issue for a long time.

I think this may be related to power management.  Are you running on a laptop?
Or are you using a virtual machine?

As a first step I would recommend upgrading to the 2017.02 release for both firmware and host software.

jpetrescu, 2017-11-09:
Thanks. I'll try the firmware update first.

closed 2019-11-27 by miek