greatscottgadgets/hackrf #404: Is HackRF dead? what else can I try?

tonyfromjupiter, 2017-09-12:
### Steps to reproduce
1. plug in HackRF usb cable to Ubuntu 16.04 on x86_64 CPU
2. do tail -F  /var/log/syslog  ;then
3. run hackrf_info

### Expected behaviour
Tell us what you expect should happen

"tail" screen should display detection of HackRF and printout various stuff

hackrf_info should display stuff

### Actual behaviour
Tell us what happens instead

No output from tail -F... 
and
tony@se2:~$ hackrf_info
No HackRF boards found.

### Version information
**Operating system**:
Linux se2 4.10.0-33-generic #37~16.04.1-Ubuntu SMP Fri Aug 11 14:07:24 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

**hackrf_info output:**

tony@se2:~$ hackrf_info
No HackRF boards found.

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
HackRF leds look OK: 3v3 green, 1v8 yellow, RF red (am I transmitting continuously)
reset button clears 1v8
reset+DFU clears DFU, stays clear

same USB slot finds Etonic RTL and gnuradio runs a simple NBFM receive pgm (built by me)

Insert any commandline or build output here
```

dominicgs, 2017-09-12:
Is there any output from 'dmesg' when you plug the HackRF in?
Which LEDs are on?

What happens if you reset your HackRF while holding down the DFU button?  (hold DFU and push reset button)

tonyfromjupiter, 2017-09-12:
no output from dmesg, as with tail -F /var/log/syslog.
HackRF leds look OK: 3v3 green, 1v8 yellow, RF red is on constantly (dims on reset button)
hold DFU and push reset button): seems to enter the "ready for DFU state" ;clears DFU, stays clear

dominicgs, 2017-09-12:
It sounds like it's not communicating over USB in the normal firmware mode.  In DFU mode do you see a device listed in dmesg?

If so, could you try to use `dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu` to write a temporary firmware to RAM on the HackRF?

tonyfromjupiter, 2017-09-13:
dmesg does not show the HackRF being inserted or removed. lsusb does not show the HackRF.

If as you suggest, I do:
tony@lenovo:~/gnuradio/hackrf/hackrf-bin-2017.02.1/firmware-bin$ dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu

I get:
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
dfu-util: No DFU capable USB device available

It looks to me as if the HackRF USB port is not working.

dominicgs, 2017-09-13:
I agree.  Could you contact the place that you bought the HackRF for a replacement?

tonyfromjupiter, 2017-09-14:
good advice.  tnx

closed 2017-09-14 by tonyfromjupiter