greatscottgadgets/hackrf #405: compiling libhackrf with MinGW 64 on linux

veso266, 2017-09-13:
Hi can someone please tell me how can I compile libhackrf (for linking it into other project) with MinGW on linux 

I am using Fedora 64 64 Bit and I cloned the repo and then tried to build as instructions for MinGW told (mingw64-cmake .. -DLIBFFTW_INCLUDE_DIR=/home/mitja/hacktv/libs/libusb-1.0.21/ (which fixed libUSB error but it still can't find FFTW
```bash
CMake Error at /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:137 (message):
  Could NOT find FFTW (missing: FFTW_LIBRARIES)
Call Stack (most recent call first):
  /usr/share/cmake/Modules/FindPackageHandleStandardArgs.cmake:377 (_FPHSA_FAILURE_MESSAGE)
  cmake/modules/FindFFTW.cmake:28 (find_package_handle_standard_args)
  hackrf-tools/src/CMakeLists.txt:26 (find_package)

-- Configuring incomplete, errors occurred!
See also "/home/mitja/hacktv/libs/hackrf/host/build/CMakeFiles/CMakeOutput.log".
See also "/home/mitja/hacktv/libs/hackrf/host/build/CMakeFiles/CMakeError.log".
```
Thanks for Anwsering and Best Regards
PS: can anyone send me prebuild dll for 64 bit windows to link them against another project?

dominicgs, 2017-09-13:
There are some instructions for building withMinGW in the [readme file](https://github.com/mossmann/hackrf/tree/master/host#for-mingw) but I haven't seenanyone build hackrf_sweep (the tool that requires FFTW3) usingMinGW.  I think it may work with some of the cmake config from the Visual Studio instructions.

The [PothosSDR installers](http://downloads.myriadrf.org/builds/PothosSDR/?C=M;O=D) include binaries for libhackrf and the HackRF tools.

veso266, 2017-09-15:
I've secsessfuly compile latest Libhackrf with MinGW
mingw64-cmake ../ -DLIBUSB_INCLUDE_DIR=/home/mitja/hacktv/knjiznice/libusb-1.0.21/libusb -DFFTW_LIBRARIES=/home/mitja/hacktv/knjiznice/hackrf_latest/FFTW/fftw (precompiled FFTW is here: http://fftw.org/install/windows.html)

dominicgs, 2017-09-15:
That's great, thanks for updating the issue.  I'm going to close it now as I believe the issue is resolved.

closed 2017-09-15 by dominicgs