greatscottgadgets/hackrf #409: I/O Error on first read or few minutes into running a scan

brazda, 2017-09-24:
### Steps to reproduce
1.run hackrf_info
2.run hackrf_info again
3.

### Expected behaviour
To receive version on first run

### Actual behaviour
With any software the first attempt to connect to the hackrf always errors out.
When running a second time it runs but will eventually error out again and loose connection.

### Version information
Antegros 

**hackrf_info output:**

### Output
```
[root@MrSmith ~]# hackrf_info 
hackrf_info version: 2017.02.1
libhackrf version: 2017.02.1 (0.5)
Found HackRF
Index: 0
hackrf_open() failed: Input/Output Error (-1000)

[root@MrSmith ~]# 
[root@MrSmith ~]# hackrf_info 
hackrf_info version: 2017.02.1
libhackrf version: 2017.02.1 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000909864c8386e7ecf
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x004c4f4e

```

dominicgs, 2017-09-24:
This is often a symptom of power saving settings. Until you try to connect
to the device the host has it powered down.

Are you running on a laptop? Do you have laptop-modes or TLP installed?

brazda, 2017-09-24:
Sounds reasonable I have notice when I run hackrf_info the usb light on the unit flashes off and back on again twice when I get the IO error but if I run hackrf_info back to back really quick the light never flashes and the command works. I am on a laptop but not to sure how to disable power management just for the usb never had to step in the world. I will do some digging but if you can recommend a course of action it will help. Thanks...

brazda, 2017-09-24:
That's defiantly the issue.. Here is what I did to check it..

`for d in /sys/bus/usb/devices/[0-9]* ; do if [[ -e $d/product ]] ; then echo -e "`basename $d`\t`cat $d/power/control`\t`cat $d/speed`\t`cat $d/product`" ; fi ; done
`
Output
1-1	auto	480	HackRF One
1-9	auto	480	HP HD Camera

`echo 'on' | tee /sys/bus/usb/devices/1-1/power/control`

Re-ran the previous command

1-1	on	480	HackRF One
1-9	auto	480	HP HD Camera

Now I seem to be able to run it reach time with no failure... 

Now time to figure out how to set it for reboots as well. If I figure out how to on Antergros I will respond.

closed 2017-09-24 by brazda

dominicgs, 2017-09-25:
This part of the FAQ may help: https://github.com/mossmann/hackrf/wiki/FAQ#power-saving-and-usb-autosuspend

I'm sorry I couldn't find it the other day, I was replying via my phone.