greatscottgadgets/hackrf #410: Hardware sync between multiple HackRFs works only the first time after reset

barto-, 2017-09-25:
### Steps to reproduce
1. Record a signal with hardware sync enabled
```sh
$ hackrf_transfer -r rx1.bin -H 1 | hackrf_transfer -r rx2.bin -H 1
```
2. Repeat step 1

3. Reset both HackRFs and repeat step 1

### Expected behaviour
The recordings should be synchronised after step 1, step 2, and step 3

### Actual behaviour
The recordings are not synchronised after step 2. Hardware synchronisation appears to be working only once after reset.
Is it possible that a buffer somewhere contains rubbish from the previous recording?

dominicgs, 2017-09-25:
Yes, it's entirely possible that we aren't resetting the buffers when we stop receiving.

barto-, 2017-09-26:
Do you have any suggestions on where I should start looking?

dominicgs, 2017-09-26:
I've had a look, and I think the only thing we don't reset when we change to `TRANSCEIVER_MODE_OFF` is the [USB bulk buffer pointer](https://github.com/mossmann/hackrf/blob/master/firmware/hackrf_usb/sgpio_isr.c#L55)

The pointer is used by the SGPIO interrupt handler to know where the samples should be written.  It's also used when scheduling the USB transfers.  It's possible that if we start with it in the second buffer, we'll transfer the first (with junk data) to the host.

I think the right thing to do is to reset the pointer when we change the transceiver state.

dominicgs, 2017-09-26:
You could try this commit: https://github.com/mossmann/hackrf/pull/408/commits/15d67a7ea8a76dbf8ec0e42c6f5f91bec6a3d77b

The branch that I've been working on has some broken functionality, so I would advise taking that commit, but not the rest of it.

barto-, 2017-09-27:
Thank you for your help @dominicgs . I've tried your commit without success. There must be something else...

dominicgs, 2017-09-27:
I'm not sure where else to look, but I'd like to find this.  It seems to me the issue is not resetting all of our state when we switch transceiver state.

martinling, 2022-03-18:
It would be good to retest this one. After all the work in #707, #982 and #1045I would be very surprised if there is anything still wrong.

mossmann, 2022-09-15:
This is indeed fixed. Thanks, @martinling!

closed 2022-09-15 by mossmann