greatscottgadgets/hackrf #411: does HackRF support usb-C?

shanggdlk, 2017-09-27:
I'm wondering whether I can use a USB-adapter to connect the HackRF USB-A port to the new Macbook USB-C port?

kpreid, 2017-09-27:
* USB C is backwards compatible. You don't need to worry about whether individual devices support it as long as you have a standard port and adapter cable.
* The HackRF One has a *B* port, not an A port, as it should.

(And I can also confirm by testing that the HackRF One will work with USB C Android devices. I don't have a new MacBook handy.)

shanggdlk, 2017-09-27:
Thanks, appreciated!

Dygear, 2017-09-27:
Via the adapter it does yes.

dominicgs, 2017-09-27:
As @kpreid and @Dygear have said, HackRF works with USB-C ports with the appropriate adapter.  I can confirm that I often use mine via the USB-C port on my laptop.

I'm going to close this issue.

closed 2017-09-27 by dominicgs

chudgoo, 2017-09-27:
I've successfully run two units off one Thunderbolt 3 port on a MacBook Pro
15" (TB) with a USB-C to USB-A (USB3 hub).

When I run one at 10MHz and the other at 20MHz, things are smooth
There is a bit of stuttering in CubicSDR, but not in GQRX when both are at
20MHz each.

That said, I'd love to see a future hardware revision take advantage of
TB3's bandwidth and power delivery features.

Dygear, 2017-09-27:
It’s kind of a waste to be honest. TB3 controller chips are expensive. If anything a USB-C native version of HackRF One, perhaps called HackRF One-C would be fine. Anything in the USB 3.0 Gen 1 family would likely still be overkill offering way more bandwidth than any of the chips on the current spec HackRF One could use. It would be possible as the HackRF is “Open Hardware” to do this ourselves. But I know nothing about circuit design.

SirDeadPuppy, 2020-09-19:
can we make one how do we do it can i by a bord and instead of soder a mirco usb port i put a c port insuead like what power chips and bits do i need to change it is open souce so buy bits and soder right? but what bits change with usbc instead ?