greatscottgadgets/hackrf #412: Help Understanding hackrf_sweep binary output

ghostop14, 2017-09-29:
I'm looking to integrate the binary output from hackrf_sweep and I wanted to get a little insight into the code segment below from hackrf_sweep.c so I can incorporate the output accordingly.  So 2 questions:

1. It looks like it writes 2 fftSize/4 blocks out so wouldn't record_length be 2*sizeof(band_edge) + **fftSize/2** * sizeof(float)?  Looks like the fwrites really do output fftSize/4 twice for an fftSize/2 total record output.

2. Also, what's the reasoning behind the pwr array being output in offset chunks? The first starting at 1+(fftSize*5)/8 and the second starting at 1+(fftSize/8)?  Are they in the correct bin order but just an artifact of I/Q and the way the data comes in or are they out of order (would I need to reorder them to have a coherent FFT curve)?

Thanks!

---------------------------------------------------------------------------
		if(binary_output) {
			record_length = 2 * sizeof(band_edge)
					+ (fftSize/4) * sizeof(float);

			fwrite(&record_length, sizeof(record_length), 1, fd);
			band_edge = frequency;
			fwrite(&band_edge, sizeof(band_edge), 1, fd);
			band_edge = frequency + DEFAULT_SAMPLE_RATE_HZ / 4;
			fwrite(&band_edge, sizeof(band_edge), 1, fd);
			fwrite(&pwr[1+(fftSize*5)/8], sizeof(float), fftSize/4, fd);

			fwrite(&record_length, sizeof(record_length), 1, fd);
			band_edge = frequency + DEFAULT_SAMPLE_RATE_HZ / 2;
			fwrite(&band_edge, sizeof(band_edge), 1, fd);
			band_edge = frequency + (DEFAULT_SAMPLE_RATE_HZ * 3) / 4;
			fwrite(&band_edge, sizeof(band_edge), 1, fd);
			fwrite(&pwr[1+fftSize/8], sizeof(float), fftSize/4, fd);
---------------------------------------------------------------------------

PhilW-WBC, 2017-11-09:
It would be nice to get a bit more information about the sweep packets.  They do seem to change quite a bit depending on how many ranges are set for the sweep.

RoninActual, 2018-12-14:
Looking forward to some more documentation on hackrf_sweep as well, very useful tool.

dominicgs, 2019-01-16:
I've added some documentation to the `hackrf_sweep` wiki page: https://github.com/mossmann/hackrf/wiki/hackrf_sweep

closed 2019-01-16 by dominicgs

kethan1, 2025-02-21:
> I've added some documentation to the `hackrf_sweep` wiki page: https://github.com/mossmann/hackrf/wiki/hackrf_sweep

I believe this link no longer functions due to a migration of the repository location.
Web Archive Link for anyone looking for it: https://web.archive.org/web/20201111195200/https://github.com/mossmann/hackrf/wiki/hackrf_sweep

kethan1, 2025-02-21:
> I've added some documentation to the `hackrf_sweep` wiki page: https://github.com/mossmann/hackrf/wiki/hackrf_sweep

New link for anyone looking for it: https://hackrf.readthedocs.io/en/latest/hackrf_tools.html#hackrf-sweep