greatscottgadgets/hackrf #414: Looking to stream recorded baseband samples upon GPIO pin high.

gtg498u, 2017-10-15:
Hi, I just wanted to know where in the firmware I could enable this function:

When a GPIO pin goes high, begin streaming the a hardcoded vector of baseband samples straight from on-chip memory to the RF mixer for transmission. The vector should be continuously streamed (looped) until the GPIO pin goes low.

I'm thinking I may need to "fake" a stream request from the usb (as if the computer was sending it) and port it to the transmit operation of the MAX chip.

dominicgs, 2017-10-16:
You can probably achieve this in the main loop in `hackrf_usb.c`.  If you set the transceiver mode to TX and put your data in the buffers, you should be able to stop the USB bulk transfers from happening and repeatedly transmit the same buffers.

gtg498u, 2017-10-17:
So I've dug into the source code more.
I see that: the usb_bulk_buffer is allocated 32768 elements, of type uint8_t.
That makes sense to me. 

But in the main function of hackrf_usb, I see that the usb_bulk_buffer is broken into two buffers.

buffer 1: [ 0 ... 16383]
buffer 0: [16384 ... 32767]

I understand that I'll be filling these buffers with my own samples. Here are my questions:

1. I'm not using the USB. I'm hard-coding the bulk buffer with my own baseband samples, I should be able to resize the bulk_buffer to the maximum array size? How large could I make the usb_bulk_buffer array length?

2. How do I fill the bulk buffer with my data?  Is buffer 0 the I data and buffer 1 the Q data?

3. How do I loop it to keep sending the same hardcoded data?

dominicgs, 2017-10-18:
1) Yes, you could resize the buffers but you'll also need to change the maximum value that the sgpio_isr code points to so that it doesn't run off the array.  I'm not sure how much spare RAM there is to expand the buffers.

2) The buffers contain interleaved samples I1,Q1,I2,Q2, etc

3) It should continuously loop if you comment out the USB transfers.

gtg498u, 2017-10-18:
Ok so here is the update...
In the main function, all of this is the same...

	usb_set_configuration_changed_cb(usb_configuration_changed);
	usb_peripheral_reset();
	
	usb_device_init(0, &usb_device);
	
	usb_queue_init(&usb_endpoint_control_out_queue);
	usb_queue_init(&usb_endpoint_control_in_queue);
	usb_queue_init(&usb_endpoint_bulk_out_queue);
	usb_queue_init(&usb_endpoint_bulk_in_queue);

	usb_endpoint_init(&usb_endpoint_control_out);
	usb_endpoint_init(&usb_endpoint_control_in);
	
	nvic_set_priority(NVIC_USB0_IRQ, 255);

	hackrf_ui_init();

	usb_run(&usb_device);

	unsigned int phase = 0;
    rf_path_init(&rf_path);
    operacake_init();

Then I start adding my portion...I begin with setting all of the transceiver settings myself...

    set_transceiver_mode(TRANSCEIVER_MODE_TX);

	uint64_t myFreq = 2400000000;	
	hackrf_ui_setFrequency(myFreq);
	
	uint32_t mySampleFreq_hz = 10000000;
	hackrf_ui_setSampleRate(mySampleFreq_hz);

	hackrf_ui_setDirection(RF_PATH_DIRECTION_TX);

	uint32_t myTXGain = 15;
	hackrf_ui_setBBTXVGAGain(myTXGain);

Next, I fill the usb_bulk_buffer completely with my 64 pt waveform (repeated 512 times)

    int i, myI;
	myI = 0;
	uint8_t myArr[] = {255,8,225,63,154,136,92,177,76,169,104,132,140,105,151,113,127,145,95,169,85,163,105,134,136,109,149,112,131,140,100,166,85,166,100,140,131,112,149,109,136,134,105,163,85,169,95,145,128,113,151,105,140,132,104,169,76,177,92,136,154,63,225,8};
	
	for (i = 0; i < 32768; ++i)
	{
	     if ((i!=0) & (i % 64 == 0))
	     {
	          myI = 0;	
	     }
  	     usb_bulk_buffer[i] = myArr[myI];
	}

I then make sure the LEDs are correct...(my favorite part of the code btw)

    led_off(LED2);
	led_on(LED3);

Then the while loop as...

    while(true) {

		// Set up IN transfer of buffer 0.
		if ( usb_bulk_buffer_offset >= 16384
		     && phase == 1
		     && transceiver_mode() != TRANSCEIVER_MODE_OFF) {
			usb_transfer_schedule_block(
				(transceiver_mode() == TRANSCEIVER_MODE_RX)
				? &usb_endpoint_bulk_in : &usb_endpoint_bulk_out,
				&usb_bulk_buffer[0x0000],
				0x4000,
				NULL, NULL
				);
			phase = 0;
		}

		// Set up IN transfer of buffer 1.
		if ( usb_bulk_buffer_offset < 16384
		     && phase == 0
		     && transceiver_mode() != TRANSCEIVER_MODE_OFF) {
			usb_transfer_schedule_block(
				(transceiver_mode() == TRANSCEIVER_MODE_RX)
				? &usb_endpoint_bulk_in : &usb_endpoint_bulk_out,
				&usb_bulk_buffer[0x4000],
				0x4000,
				NULL, NULL
			);
			phase = 1;
		}
	}

Where I removed the CPLD Update and Sweep Mode Check (I don't need those). I did not change the size of the usb_bulk_buffer (yet...)

Problem: I'm not seeing the waveform appear on my receiving HackRF. 

Verification: I verified that the same 64-pt complex baseband file that I converted to the uint8_t hardcoded vector that I set the usb_bulk_buffer block with, does actually transmit from a GNUradio flow graph. This was verified because I see my signal appear in the FFT. 

Observations: The LED3 does not remain on, like I thought it should.

Thoughts: So I'm not sure what I'm missing here. Other than, there may be a call to some: TXMIT ALL OF THE USB_BULK_BUFFER code that I can't seem to generate. Even then, will I have to keep refilling the usb_bulk_buffer array within the while loop?

dominicgs, 2017-10-18:
The LED likely gets changed by some other part of the firmware after you set it.

Some feedback on your changes:
 * I would call `set_transceiver_mode(TRANSCEIVER_MODE_TX);` last of all.
 * You're calling the `_ui_` functions, these update the UI, but don't affect the values that you're setting.
 * The while loop is checking for USB bulk transfers, but you don't have any, so I would comment out most, if not all of the content of that loop.

gtg498u, 2017-10-18:
Thank you. I'm using the actual function calls that should be changing the RF end.

uint64_t myFreq = 2400000000;	
	set_freq(myFreq);
	
	uint32_t mySampleFreq_hz = 10000000;
	sample_rate_set(mySampleFreq_hz);

	uint32_t myTXGain = 15;
	max2837_set_txvga_gain(&max2837, myTXGain);

	rf_path_set_direction(&rf_path,RF_PATH_DIRECTION_TX);
	led_off(LED2);
	led_on(LED3);

But now, I'm trying to figure out the function calls for sending my uint8_t array out as TX.

dominicgs, 2017-10-18:
The call to `set_transceiver_mode(TRANSCEIVER_MODE_TX);` should start transmitting the values from the buffer.

gtg498u, 2017-10-19:
So far so good. Managed to get the usb_bulk_buffer transmitting. However, I may not have used the correct values for the uint8_t myArr array. In MATLAB my waveform looks fine. Again, when I send the same data, but in complex binary data format as a file, through GNUradio to osmocom out through HackRF, then I see the signal correctly in the fft of the other hackrf receiving. 

Right now, I'm wondering if I'm converting the complex baseband floats correctly to uint8_t with the proper scaling. In MATLAB I generate the 64-pt complex baseband signal.

%% Begin MATLAB psuedo-code
let's call it X.
I then convert it to uint8 through:
I = uint8( real(X)*128+127.5 );
Q = uint8( imag(X)*128+127.5 );

Next I interleave:
outarr = zeros(1,128);
outarr(1:2:end) = I;
outarr(2:2:end) = Q;

So now the total length of outarr is:
128, with the format: [I1 Q1 I2 Q2 ... ]
Those raw values are what I copy and paste into my code.

Here is the final code:

    int main(void) {
	pin_setup();
	enable_1v8_power();
    #if (defined HACKRF_ONE || defined RAD1O)
	enable_rf_power();

	/* Let the voltage stabilize */
	delay(1000000);
    #endif
	cpu_clock_init();

	usb_set_descriptor_by_serial_number();

	usb_set_configuration_changed_cb(usb_configuration_changed);
	usb_peripheral_reset();
	
	usb_device_init(0, &usb_device);
	
	usb_queue_init(&usb_endpoint_control_out_queue);
	usb_queue_init(&usb_endpoint_control_in_queue);
	usb_queue_init(&usb_endpoint_bulk_out_queue);
	usb_queue_init(&usb_endpoint_bulk_in_queue);

	usb_endpoint_init(&usb_endpoint_control_out);
	usb_endpoint_init(&usb_endpoint_control_in);
	
	nvic_set_priority(NVIC_USB0_IRQ, 255);

	hackrf_ui_init();

	usb_run(&usb_device);

	rf_path_init(&rf_path);
	operacake_init();

	//unsigned int phase = 0;

	uint64_t myFreq = 2400000000;	
	set_freq(myFreq);
	
	uint32_t mySampleFreq_hz = 10000000;
	sample_rate_set(mySampleFreq_hz);

	uint32_t myTXGain = 15;
	max2837_set_txvga_gain(&max2837, myTXGain);

	rf_path_set_direction(&rf_path,RF_PATH_DIRECTION_TX);
	led_off(LED2);
	led_on(LED3);

	int i, myI;
	myI = 0;
	uint8_t myArr[] = {133,127,123,125,131,131,125,123,127,133,129,123,124,131,132,125,122,127,132,129,124,124,129,132,127,122,125,132,131,124,123,129,133,127,123,125,131,131,125,123,127,133,129,123,124,131,132,125,122,127,132,129,124,124,129,132,127,122,125,132,131,124,123,129,133,127,123,125,131,131,125,123,127,133,129,123,124,131,132,125,122,127,132,129,124,124,129,132,127,122,125,132,131,124,123,129,133,127,123,125,131,131,125,123,127,133,129,123,124,131,132,125,122,127,132,129,124,124,129,132,127,122,125,132,131,124,123,129};

	for (i = 0; i < 32768; ++i)
	{
		if ((i!=0) & ((i % 128) == 0))
		{
			myI = 0;	
		}
  		usb_bulk_buffer[i] = myArr[myI];
		++myI;
	}

    set_transceiver_mode(TRANSCEIVER_MODE_TX);
	while(true) {
    set_transceiver_mode(TRANSCEIVER_MODE_OFF);
    set_transceiver_mode(TRANSCEIVER_MODE_TX);
    delay(10000000);

	}

	return 0;
    }

I need to turn the radio off and add a processing delay in order to make the hackrf responsive to the usb calls from the terminal.

So my final question is: Am I transmitting the signal I actually want to transmit, from usb_bulk_buffer?

gtg498u, 2017-10-22:
Buffers are full with my own signal.
I'm transmitting the correct signal (I didn't realize the DAC values needed to be between 0 and 127 until i looked at the hackrf_transfer help options). 
But now my signal has a huge DC spike, that shouldn't be there...
When I transmit the same signal content from GNUradio using file source, DC spike does not appear. My receiving flowgraph removes DC content with correctiq block. Are the usb_bulk_buffer values supposed to be between 0 and 127?

dominicgs, 2017-10-27:
The values should be between -128 and 127.  This is probably why you see a large DC spike.

gtg498u, 2017-12-19:
Sorry for not updating the forums earlier. But here is the code to accomplish what I wanted. Thanks @dominicgs for the guidance. 

Insert this new function before the main loop
```
void reloadbuffer(uint8_t* usb_bulk_buffer, float* newData){
   memset(usb_bulk_buffer,0,32768);
	int i, myI;
	myI = 0;
	for (i = 0; i < 32768; ++i) //32768
	{
		if((i%128)==0){
			myI = 0;
		}
		usb_bulk_buffer[i] = (int8_t)(128.0f*99.9f*newData[myI]);		
		++myI;
	}
}
```

Then here is the code for the main loop
```
int main(void) {
	pin_setup();
	enable_1v8_power();
#if (defined HACKRF_ONE || defined RAD1O)
	enable_rf_power();

	/* Let the voltage stabilize */
	delay(1000000);
#endif
	cpu_clock_init();

	usb_set_descriptor_by_serial_number();

	usb_set_configuration_changed_cb(usb_configuration_changed);
	usb_peripheral_reset();
	
	usb_device_init(0, &usb_device);
	
	usb_queue_init(&usb_endpoint_control_out_queue);
	usb_queue_init(&usb_endpoint_control_in_queue);
	usb_queue_init(&usb_endpoint_bulk_out_queue);
	usb_queue_init(&usb_endpoint_bulk_in_queue);

	usb_endpoint_init(&usb_endpoint_control_out);
	usb_endpoint_init(&usb_endpoint_control_in);
	
	nvic_set_priority(NVIC_USB0_IRQ, 255);

	hackrf_ui_init();

	usb_run(&usb_device);

	rf_path_init(&rf_path);
	operacake_init();

	//unsigned int phase = 0;

	uint64_t myFreq = 2400000000;	
	set_freq(myFreq);
	
	uint32_t mySampleFreq_hz = 10000000;
	sample_rate_set(mySampleFreq_hz);

	uint32_t myTXGain = 25;
	max2837_set_txvga_gain(&max2837, myTXGain);

	float myArr2[] = {0.010000,0.000000,-0.009546,0.001444,0.008235,-0.002724,-0.006207,0.003699,0.003681,-0.004268,-0.000922,0.004388,-0.001787,-0.004077,0.004179,0.003409,-0.006036,-0.002500,0.007207,0.001495,-0.007628,-0.000542,0.007322,-0.000231,-0.006387,0.000732,0.004987,-0.000921,-0.003320,0.000811,0.001594,-0.000469,0.000000,0.000000,-0.001309,0.000469,0.002236,-0.000811,-0.002743,0.000921,0.002852,-0.000732,-0.002630,0.000231,0.002179,0.000542,-0.001612,-0.001495,0.001036,0.002500,-0.000535,-0.003409,0.000164,0.004077,0.000058,-0.004388,-0.000146,0.004268,0.000137,-0.003699,-0.000080,0.002724,0.000023,-0.001444,0.000000,-0.000000,0.000023,0.001444,-0.000080,-0.002724,0.000137,0.003699,-0.000146,-0.004268,0.000058,0.004388,0.000164,-0.004077,-0.000535,0.003409,0.001036,-0.002500,-0.001612,0.001495,0.002179,-0.000542,-0.002630,-0.000231,0.002852,0.000732,-0.002743,-0.000921,0.002236,0.000811,-0.001309,-0.000469,-0.000000,0.000000,0.001594,0.000469,-0.003320,-0.000811,0.004987,0.000921,-0.006387,-0.000732,0.007322,0.000231,-0.007628,0.000542,0.007207,-0.001495,-0.006036,0.002500,0.004179,-0.003409,-0.001787,0.004077,-0.000922,-0.004388,0.003681,0.004268,-0.006207,-0.003699,0.008235,0.002724,-0.009546,-0.001444};

	float myArr71[] = {0.010000,0.000000,-0.009546,0.001444,0.008235,-0.002724,-0.006207,0.003699,0.003681,-0.004268,-0.000922,0.004388,-0.001787,-0.004077,0.004179,0.003409,-0.006036,-0.002500,0.007207,0.001495,-0.007628,-0.000542,0.007322,-0.000231,-0.006387,0.000732,0.004987,-0.000921,-0.003320,0.000811,0.001594,-0.000469,0.000000,0.000000,-0.001309,0.000469,0.002236,-0.000811,-0.002743,0.000921,0.002852,-0.000732,-0.002630,0.000231,0.002179,0.000542,-0.001612,-0.001495,0.001036,0.002500,-0.000535,-0.003409,0.000164,0.004077,0.000058,-0.004388,-0.000146,0.004268,0.000137,-0.003699,-0.000080,0.002724,0.000023,-0.001444,0.000000,-0.000000,0.000023,0.001444,-0.000080,-0.002724,0.000137,0.003699,-0.000146,-0.004268,0.000058,0.004388,0.000164,-0.004077,-0.000535,0.003409,0.001036,-0.002500,-0.001612,0.001495,0.002179,-0.000542,-0.002630,-0.000231,0.002852,0.000732,-0.002743,-0.000921,0.002236,0.000811,-0.001309,-0.000469,-0.000000,0.000000,0.001594,0.000469,-0.003320,-0.000811,0.004987,0.000921,-0.006387,-0.000732,0.007322,0.000231,-0.007628,0.000542,0.007207,-0.001495,-0.006036,0.002500,0.004179,-0.003409,-0.001787,0.004077,-0.000922,-0.004388,0.003681,0.004268,-0.006207,-0.003699,0.008235,0.002724,-0.009546,-0.001444};
[... 4 lines cut ...]
	
	enable_gpio3_8_input();
	
	bool gpio3_8_val = 0;
	while(true) { 	
	
	gpio3_8_val = read_gpio3_8();
	if(gpio3_8_val){
	reloadbuffer(usb_bulk_buffer,(float*)myArr2);
	set_transceiver_mode(TRANSCEIVER_MODE_OFF);
	set_transceiver_mode(TRANSCEIVER_MODE_TX);
	delay(1800000);	
	}else{
	reloadbuffer(usb_bulk_buffer,(float*)myArr128);
	set_transceiver_mode(TRANSCEIVER_MODE_OFF);
	set_transceiver_mode(TRANSCEIVER_MODE_TX);
	delay(1800000);
	}
	
	}

	return 0;
}
```
The code above will oscillate between sending two different OFDM signals. It's a nice compact way to send a single signal. I'll post the high trigger line later.