greatscottgadgets/hackrf #415: Trouble updating Firmware

steven-bloom, 2017-10-18:
Hi everyone
I wanted to update Firmware for my Hackrf, but facing that Problem. Is that normal behavior
[image]
I really appreciate any help you can provide

dominicgs, 2017-10-18:
Yes, that's how a normal firmware update looks.  I'm aware that it's not entirely clear that it succeeded.

I can add a line to make it more obvious that it's complete, but we don't know if the write was successful until you reset the device and run the new firmware.

steven-bloom, 2017-10-18:
Hello, Dominicgs! Thanks for your response. 
I did reset. The output of the command "hackrf_info" is below.
[image]
The problem is, however, that these hackrf does not transmit output signal correct.
I have two HackRFs and one GNU-Radio Flowgraph that produce DRM-Radio Signal. One the Picture below is one of my HackRF that does that proper (wide signal)
[image]
 Аnd in this picture  of the same signal, but produced with another HackRF (with the topic in question). No changes have been made, only another HackRF. And it can not be properly decoded.
Thats why i decided to make Firmware update 
[image]
Anyway, I'm still getting this wrong narrow signal even with Firmware-update.
Assuming that Firmware is correct installed, what could be the problem
Thanks for help again

gtg498u, 2017-10-18:
Try putting a Correct IQ block right after the osmocom source block.
[image]

dominicgs, 2017-10-18:
Is the CPLD up to date?  You can update it with `hackrf_cpldjtag -x hackrf_cpld_default.xsvf` from the`firmware-bin` directory.

Instead of transmitting from GNU Radio, could you use a file sink (signed bytes format) and transmit it with `hackrf_transfer` using the two radios.  This helps me to narrow down where the problem is.

steven-bloom, 2017-10-19:
It didn't worked, I tried with hackrf_transmit and i can not still receive wideband signal (Picture 1) unlike to Picture 2 
[image]
[image]

I have also CPLD updated, and it still didnt  worked
I also tried Firmware 2015...no result

hustpigeon, 2017-10-23:
I have the problem like him， before I update newest firmware, at least I can receive some FM, and now none can receive

hustpigeon, 2017-10-24:
I solved this problem by update CPLD firmware

steven-bloom, 2017-10-24:
CPLD is uptodate,  but it stil doesnt work :(

dominicgs, 2017-11-02:
@steven-bloom Could you contact your reseller for a replacement HackRF?  I think we need to get this one back to analyse it for defects.

closed 2018-04-05 by dominicgs