greatscottgadgets/hackrf #416: HackRF One for Ranging/Localization Application

elal007, 2017-10-21:
### Steps to reproduce
1. Investigate the possibility of using HackRF for localization. And how accurate it is?. 
2. Using RSSI method, we can see from hackrf_sweep that we can produce the RSS. I just curious, how can we get the rssi from specific transmitter (Tx). Example Tx and Rx both work on 2412 GHz, and we need three Tx to estimate the unknown position of Rx. How can we differentiate the signal strength from each transmitter (Tx). I ever make the same one with Tmote Sky Node with easy.
3.If we want to use TDoA, then we need to synchronize the HackRF. What is the time resolution? From my understanding, it should have 15 Meter resolution. It was based on 10Mhz clock of si5351c.
4. If we use ToF, then we didnot need synchronization, but we need to timestamp the message sending from Rx to Tx, back from Tx Rx. Does anybody ever try the possibility using this method?

### Expected behavior
I want to estimate the distance between a transmitter (Tx) and receiver (Rx). A number of transmitters are minimum 3. 

### Actual behaviour
How can we

### Version information
**Operating system**:
Ubuntu 16.04
**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
How can we use GRC to process I/Q signal so that we can differentiate the RSSI from each transmitter. Or calculate the time of flight message from Tx to RX.

gtg498u, 2017-10-23:
Perhaps you could do this:

1. Use an OFDM short sequence preamble structure, with your own OFDM symbol appended which encodes which transmitter sent the packet.

2. Take the SNR for the preamble as your RSS.

3. Place the transmitters at specific locations, have them broadcast continuously at set intervals (perhaps space their broadcast rate out at different intervals). As your receiver moves through the point space, calculate the RSS from each transmitter. Correlate with eastimatwd point in space, with a calibrated look-up table from a priori measurements of the area.

elal007, 2017-10-23:
Hi gtg498u,
Thank you for your response. I am new with data communication. I've been reading some material about ofdm structure to know how to make it. I read paper "IEEE 802.11 OFDM Software Defined Radio
Beacon Frame Transmission", I hope it will work with my plan. Please give me suggestion, if you have other resource, plz.

barto-, 2017-10-23:
I've used TDOA successfully. For synchronisation take a look here #381 .
The accuracy you can achieve is +- 15m.

gtg498u, 2017-10-24:
So, you could make the signal for each transmitter in MATLAB or Octave, and their flowgraph would simply be transmitting from a file. Let me know if you would like to collaborate. You can do all the processing off-line.

elal007, 2017-10-25:
@barto- @gtg498u ,
Thanks for your guides. It's really help me. Currently, I need to make sure the signals is getting from the transmitter. May be I need to transmit the signal with transmitter id information. I didnot use Matlab for experiment. I am using GRC. Can you guide me, how can i use GNU Radio to get the signal with id transmitter information?.  Later, I also want to develop the standalone hackrf, which is means that the hackrf will transmit the signal when it's power on. So, I didnot need to give a command hackrf_transfer. Can somebody help me, how to make it?. 

Thanks in advance

gtg498u, 2017-10-27:
Sure! We can collaborate. Maybe talking offline would be better?

gtg498u, 2017-10-29:
@elal007 Did you still need help?

elal007, 2017-10-29:
Hi @gtg498u ,
Yes, sure. I am still working on it. I send you message to ur mail. Please reply it, so we can collaborate.

Thx in advance

gtg498u, 2017-10-30:
Check my profile page

elal007, 2017-10-30:
Hi @gtg498u ,
I send you mail.

thx

dominicgs, 2017-11-16:
On the assumption that the two of you are now communicating via email, and that this isn't something that will require modifications to HackRF software or firmware for now, I am closing this issue.

closed 2017-11-16 by dominicgs