greatscottgadgets/hackrf #417: libhackrf does not install using provided instructions

nickspurry, 2017-10-22:
### Steps to reproduce
1. Download latest repository or release
2. Compile and install using provided instructions
a. mkdir hostébuild
b. cd host/build
c. cmake ..
d. make
e. sudo make install
f. sudo ldconfig
3. Run hackrf_info

### Expected behaviour
I would expect that the hackrf_tools would work from the command line after installation.  

### Actual behaviour
hackrf_tools work fine when specifically using the versions in the 'build' folder but not when using the installed versions.

### Version information
**Operating system**:
Raspbian running on Raspberry Pi 3

**hackrf_info output:**
```
hackrf_info version: 2017.02.1
hackrf_info symbol lookup error: hackrf_info: undefined symbol: hackrf_library_release
```

dominicgs, 2017-10-22:
It looks like you have both the latest installed version and the package from Raspian installed.  Try running `sudo apt-get remove --purge libhackrf0 hackrf`?

nickspurry, 2017-10-22:
This appears to have fixed the issue. I had previously removed both but without the --purge flag. Thanks for responding so quickly. On a side note, how would one go about updating the package in Raspbian and possibly other platforms? It seems terribly out of date.

dominicgs, 2017-10-22:
We don't produce those packages, we have to wait for the package maintainers to update them.

closed 2017-11-02 by dominicgs