greatscottgadgets/hackrf #418: can hackrf one support OpenBTS?

hustpigeon, 2017-10-24:
### Steps to reproduce
1.
2.
3.

### Expected behaviour
Tell us what you expect should happen

### Actual behaviour
Tell us what happens instead

### Version information
**Operating system**:

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

hustpigeon, 2017-10-31:
no one answer?

dominicgs, 2017-11-02:
There are two stumbling blocks to using OpenBTS with HackRF One
 * HackRF One is a half duplex transceiver, whereas GSM is a full duplex protocol
 * The up/downlink frequencies for GSM are 20MHz apart, which is at the maximum bandwidth of HackRF One

I have heard of some people having success with two HackRFs or using a HackRF for transmit and an RTL-SDR for receive, but I haven't got a link to a guide or any documentation.

hustpigeon, 2017-11-02:
@dominicgs thank you very much sir! you are so nice!

closed 2017-11-02 by dominicgs

joskar201, 2025-01-14:
Hi everyone,

I wanted to follow up on this discussion. Are there any updates or guides available for using a HackRF One and RTL-SDR combination to work with OpenBTS? Any insights or pointers would be greatly appreciated!

Thanks in advance!