greatscottgadgets/hackrf #420: HackRF has no USB Light and Not Visible on any Computer

AlexL410, 2017-10-29:
### Steps to reproduce
1. Plug in HackRF to USB Port
2. No Sound in Windows on multiple Computers
3. USB light on HackRF does not light

### Expected behaviour
Windows should recognize when HackRF is plugged in and beep

### Actual behaviour
No sound, and Zadig does not see the device to allow me to configure the drivers. USB port is not physically damaged. There is no physical damage to USB port.

### Version information
Windows 7 64 Bit and Windows 10 64Bit

**hackrf_info output:**

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

### Output
```
Insert any commandline or build output here
```

gtg498u, 2017-10-29:
I had an issue with different USB cables. Be sure you are using a short one with a ferrite choke.

dominicgs, 2017-10-29:
Is there anything listed in the device manager?

If you hold down the "DFU" button as you connect it, does a device appear?

AlexL410, 2017-10-30:
Hi gtg498u,

It looks like it was the cable. I tried about 5 of them I have, but only one worked. I do not have one with a coil on it. But at least I can access the device now. Thank you.

AlexL410, 2017-10-30:
Dominic, 
I have it working now with a different USB cable. Several others did not work.
With the non-working cables, holding the DFU button did not make the HackRF show up in device manager. It is interesting that the same cables work fine with other microcontroller micro-usb devices.
Thank you for your help.

dominicgs, 2017-10-30:
On rare occasions the micro USB ports can become a little loose.  With the non-working cables, is there some play in the connector?

AlexL410, 2017-10-30:
Dominic, 
the USB connector is not loose at all. It has a good solid mounting to the inside circuit board. There is no side to side movement at all. I tried moving between the 5 cables, and the only connection that shows up is with the working cable. Since the usb-micro connector can be broken easily, I am gentle with the insertions and removals of all my devices.
Cheers!

hustpigeon, 2017-10-31:
bad cable has week current,so hackrf one can't work correct

closed 2017-11-02 by dominicgs