greatscottgadgets/hackrf #421: who can explain this code ,thanks

hustpigeon, 2017-10-31:
in file tuning.c 

**line 78:** 
max2837_freq_nominal_hz = 2650000000 - **(freq / 7);**

why minus freq/7 ?

hustpigeon, 2017-10-31:
i get it:

for  IF is graduated from 2650 MHz to 2343 MHz 
freq's max value is 2150M(line 71)，so when 2650M-2150/7=2343, that mean freq is 0~~2150,  the IF is 2650-2343， maybe 7 is calc by this way ,of you can use other means that made it the same

         if(freq_mhz < MAX_LP_FREQ_MHZ) // 2150M
	{
		rf_path_set_filter(&rf_path, RF_PATH_FILTER_LOW_PASS);
#ifdef RAD1O
		max2837_freq_nominal_hz = 2300000000;
#else
		/* IF is graduated from 2650 MHz to 2343 MHz */
		max2837_freq_nominal_hz = 2650000000 - (freq / 7);
#endif
		mixer_freq_mhz = (max2837_freq_nominal_hz / FREQ_ONE_MHZ) + freq_mhz;
		/* Set Freq and read real freq */
		real_mixer_freq_hz = mixer_set_frequency(&mixer, mixer_freq_mhz);
		max2837_set_frequency(&max2837, real_mixer_freq_hz - freq);
		sgpio_cpld_stream_rx_set_q_invert(&sgpio_config, 1);
	}

dominicgs, 2017-11-02:
I'm glad you worked it out, I'm closing this issue as it appears to be resolved.

closed 2017-11-02 by dominicgs