greatscottgadgets/hackrf #423: why cpld code  adc_data_i xor 0x80   0x7F, what is this mean ,help me

hustpigeon, 2017-11-04:
process(host_clk_i)
    begin
        if rising_edge(host_clk_i) then
            if codec_clk_i = '1' then
                -- I: non-inverted between MAX2837 and MAX5864
                data_to_host_o <= **adc_data_i xor X"80";**
            else
                -- Q: inverted between MAX2837 and MAX5864
                data_to_host_o <= **adc_data_i xor rx_q_invert_mask;**
            end if;
        end if;
end process;

hustpigeon, 2017-11-05:
who can help me to explain ?

hustpigeon, 2017-11-05:
i have some idea,but i don't know wether it's true

dominicgs, 2017-11-16:
This is to change the values to counteract the reversal of the differential signal lines coming from the RF frontend - as discussed in #424

closed 2017-11-16 by dominicgs