greatscottgadgets/hackrf #424: why RXBBQ+ connect to QA- (why not QA+) --in hardware shecmatic.pdf

hustpigeon, 2017-11-06:
[image]

dominicgs, 2017-11-06:
I just posted this answer to a similar question on the mailing list: Yes, this was intentional.  If you look at the layout of the board, the pins are in different orders on the two parts, so routing them would involve crossing the signals, and using vias to move them on to different board layers.  All of this is a negative when it comes to signal integrity.

As they are differential signals, we can switch them as long as we correct the values in the digital domain, which we do in the CPLD here: https://github.com/mossmann/hackrf/blob/master/firmware/cpld/sgpio_if/top.vhd#L117

This is also explains your question in #423

hustpigeon, 2017-11-07:
thank you very much @dominicgs ！ i have another question about "Q  invert " why  Q invert not  I invert，
why xor 0x7f , any material or some doc help me to understand.

closed 2017-11-07 by hustpigeon