greatscottgadgets/hackrf #425: Two HackRF One in One Laptop for TX and RX

elal007, 2017-11-06:
### Steps to reproduce
1. Plug in Two HackRFOne to two ports USB in Laptop
2. Run hackrf_sweep (RX) on one Hack RF, run hackrf_transfer (TX)

### Expected behavior
I should run both from a different terminal.

### Actual behavior
I cannot run both. 

RX (hackrf_sweep)
Exiting...
Total sweeps: 134 in 0.34345 seconds (390.18 sweeps/second)
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
.01
2017-11-06, 23:14:06.415391, 2427000000, 2432000000, 1000000.00, 20, -58.90, -63.17, -71.43, -71.40, -66.16
2017-11-06, 23:14:06.417848, 2412000000, 2417000000, 1000000.00, 20, -65.65, -66.44, -69.70, -66.43, -74.02
2017-11-06, 23:14:06.417848, 2422000000, 2427000000, 1000000.00, 20, -63.54, -77.74, -64.23, -67.54, -71.09
2017-11-06, 23:14:06.417848, 2417000000, 2422000000, 1000000.00, 20, -66.29, -74.15, -71.94, -68.82, -71.29
2017-11-06, 23:14:06.417848, 2427000000, 2432000000, 1000000.00, 20, -64.75, -67.08, -66.13, -65.26, -68.37
2017-11-06, 23:14:06.420306, 2412000000, 2417000000, 1000000.00, 20, -64.41, -66.43, -76.74, -70.10, -61.93
2017-11-06, 23:14:06.420306, 2422000000, 2427000000, 1000000.00, 20, -62.97, -65.46, -65.28, -63.21, -67.00
2017-11-06, 23:14:06.420306, 2417000000, 2422000000, 1000000.00, 20, -66.81, -63.19, -71.38, -74.75, -69.84
2017-11-06, 23:14:06.420306, 2427000000, 2432000000, 1000000.00, 20, -93.39, -64.88, -63.68, -69.73, -63.23
fclose(fd) done
exit
elal@ubuntu:~/Downloads/hackrf-2017.02.1/firmware-bin$ hackrf_sweep -f 2412:2422

TX (hackrf_transfer)
elal@ubuntu:~/sdr-utils$ hackrf_transfer -d 0000000000000000925866e62d5e7ae3 -f 2412000000 -t tx-signal.dat
call hackrf_set_sample_rate(10000000 Hz/10.000 MHz)
call hackrf_set_hw_sync_mode(0)
hackrf_set_hw_sync_mode() failed: feature not supported by installed firmware (-1005)
elal@ubuntu:~/sdr-utils$ 

### Version information
**Operating system**:
Ubunty 16.04
**hackrf_info output:**
elal@ubuntu:~/Downloads/hackrf-2017.02.1/firmware-bin$ hackrf_info
hackrf_info version: git-2174aa0
libhackrf version: git-2174aa0 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000925866e62b8413e3
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x006f4f5c

Found HackRF
Index: 1
Serial number: 0000000000000000925866e62d5e7ae3
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2 (API:1.00)
Part ID Number: 0xa000cb3c 0x00794763
elal@ubuntu:~/Downloads/hackrf-2017.02.1/firmware-bin$

dominicgs, 2017-11-07:
This is due to the host tools being from the 2017.02 release and the
firmware on the second HackRF being from 2015.07.

Update the firmware on the second HackRF and it should work.

elal007, 2017-11-07:
@dominicgs ,
Sorry didnot realize that one hackrf still not upgraded to a new firmware. Thanks. It works now.

dominicgs, 2017-11-07:
No worries, I'm glad it's working now.

closed 2017-11-07 by dominicgs