greatscottgadgets/hackrf #426: Hack RG Conformal coating on pads for RF Shield

ncoop23, 2017-11-07:
There is conformal coating over the ground pads for the RF shield.  This makes it very hard to install a RF shield on new boards.

yhetti, 2017-11-07:
Where did you buy your HackRF One from? I ask because Great Scott Gadgets does not use a conformal coating on any of it's products. I have found that sometimes it is difficult to get solder to stick to the first pad. This is because it is hard to get the pad to heat evenly with the shield frame. Heat is dissipating through the ground plain and the many nearby pads are acting like a heat sink.

ncoop23, 2017-11-07:
Got it from Amazon.  When I was heating the pads, there was a lot of melting of the conformal coating which then got all over and made the solder not want to stick.  My RF shield is not attached properly now and I'm having issues with noise.  I'm about to cut it off and deal with the noise.  I have a metal case and portapack which helps a lot.

yhetti, 2017-11-09:
Can you attach a photo of your board?

ncoop23, 2017-11-09:
See attached. 
[image]

ncoop23, 2017-11-09:
I'm not the best at soldering but every time I touched the pad, a river of coating flowed in to the pad and the solder wouldn't stick. I figured whatever and went for broke but that didn't work either. I'm going to snip it off and forget about it I think.

ncoop23, 2017-11-09:
The extra header went on nicely on the back.

yhetti, 2017-11-09:
It looks to me like you have enough of it soldered to make the shield effective. If you take the pick and place bridge off of the top you may get better access to the pads near the P28 header. If you decide to solder anymore of it, make sure to use an excessive amount of flux. Then snap the shield on and you should see a noise reduction.

KC5CQW, 2018-01-16:
Actually, it looks like too much flux is there...
From the picture, I cannot find any evidence of conformal coat. Are you referring to the green solder mask?  I can say that your iron is undersized (low wattage) and not properly heating the pads on the PCB. There is already more than enough solder and flux there for a reheat if you haven’t removed anything yet. I recommend getting a 35-40watt iron and reflow the solder to ensure a good electromechanical bond. Next, clean up the PCB with rubbing alcohol (IPA) and q-tips until no flux is within the shielded RF area (at a minimum). Have no worries about cleaning the board, you won’t harm anything unless you power it up whilst still wet.

closed 2019-12-09 by miek