greatscottgadgets/hackrf #430: Building Blinky firmware

elal007, 2017-11-14:
### Steps to reproduce
I Follow instruction on https://github.com/sharebrained/portapack-hackrf/wiki/Building-Firmware
mkdir build
cd build
cmake ..

### Expected behaviour
I want to make a blinky firmware. 

### Actual behaviour
elal@ubuntu:~/hackrf/firmware/blinky/build$ make firmware
make: *** No rule to make target 'firmware'.  Stop.

### Version information
**Operating system**:
UBuntu 16.04
**hackrf_info output:**
elal@ubuntu:~/hackrf/firmware/blinky/build$ hackrf_info
hackrf_info version: git-2174aa0
libhackrf version: git-2174aa0 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000925866e62d666ee3
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00694f5d

### Output
```
elal@ubuntu:~/hackrf/firmware/blinky$ mkdir build
elal@ubuntu:~/hackrf/firmware/blinky$ cd build/
elal@ubuntu:~/hackrf/firmware/blinky/build$ cmake ..
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/lib/arm-none-eabi
-- Cross-compiling with the gcc-arm-embedded toolchain
-- Toolchain prefix: /usr/lib/arm-none-eabi
-- The ASM compiler identification is GNU
-- Found assembler: /usr/bin/arm-none-eabi-gcc
-- DFU utils version: 0.8
-- Configuring done
-- Generating done
-- Build files have been written to: /home/elal/hackrf/firmware/blinky/build
```
List of file/directory on build/CMakeFiles
elal@ubuntu:~/hackrf/firmware/blinky/build/CMakeFiles$ ls
3.5.1               blinky.dfu.dir      blinky_m0.bin.dir   blinky-program.dir               CMakeOutput.log      Makefile.cmake
blinky.bin.dir      blinky_dfu.elf.dir  blinky_m0.elf.dir   cmake.check_cache                CMakeRuleHashes.txt  progress.marks
blinky_dfu.bin.dir  blinky.elf.dir      blinky_objects.dir  CMakeDirectoryInformation.cmake  Makefile2            TargetDirectories.txt
elal@ubuntu:~/hackrf/firmware/blinky/build/CMakeFiles$ 

Please Help me.

dominicgs, 2017-11-14:
The instructions on that wiki page were a little unclear.  I've updated them.

You should run `make` or `make blinky` (they are equivalent) rather than `make firmware`

elal007, 2017-11-14:
@dominicgs 
Horeee...It works. Thank you very much.
elal@ubuntu:~/hackrf/firmware/blinky/build$ make
Scanning dependencies of target blinky_m0.elf
[  4%] Building C object CMakeFiles/blinky_m0.elf.dir/home/elal/hackrf/firmware/common/m0_sleep.c.obj
[  8%] Linking C executable blinky_m0.elf
[  8%] Built target blinky_m0.elf
Scanning dependencies of target blinky_m0.bin
[  8%] Built target blinky_m0.bin
Scanning dependencies of target blinky_objects
[ 12%] Building C object CMakeFiles/blinky_objects.dir/blinky.c.obj
[ 16%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/hackrf_core.c.obj
[ 20%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/sgpio.c.obj
[ 25%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/rf_path.c.obj
[ 29%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/si5351c.c.obj
[ 33%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/max2837.c.obj
[ 37%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/max2837_target.c.obj
[ 41%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/max5864.c.obj
[ 45%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/max5864_target.c.obj
[ 50%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/mixer.c.obj
[ 54%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/i2c_bus.c.obj
[ 58%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/i2c_lpc.c.obj
[ 62%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/w25q80bv.c.obj
[ 66%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/w25q80bv_target.c.obj
[ 70%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/spi_bus.c.obj
[ 75%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/spi_ssp.c.obj
[ 79%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/gpio_lpc.c.obj
[ 83%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/rffc5071.c.obj
[ 87%] Building C object CMakeFiles/blinky_objects.dir/home/elal/hackrf/firmware/common/rffc5071_spi.c.obj
[ 91%] Building ASM object CMakeFiles/blinky_objects.dir/m0_bin.s.obj
[ 91%] Built target blinky_objects
Scanning dependencies of target blinky.elf
[ 95%] Linking C executable blinky.elf
[ 95%] Built target blinky.elf
Scanning dependencies of target blinky.bin
[ 95%] Built target blinky.bin
Scanning dependencies of target blinky_dfu.elf
[100%] Linking C executable blinky_dfu.elf
[100%] Built target blinky_dfu.elf
Scanning dependencies of target blinky_dfu.bin
[100%] Built target blinky_dfu.bin
Scanning dependencies of target blinky.dfu
dfu-suffix (dfu-util) 0.8

Copyright 2011-2012 Stefan Schmidt, 2013-2014 Tormod Volden
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Suffix successfully added to file
dfu-prefix (dfu-util) 0.8

Copyright 2011-2012 Stefan Schmidt, 2014 Uwe Bonnes
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

Adding prefix to file
[100%] Built target blinky.dfu
elal@ubuntu:~/hackrf/firmware/blinky/build$ ls
blinky.bin  blinky_dfu.bin  blinky.elf     blinky_m0.elf   CMakeFiles           m0_bin.s  m4.map
blinky.dfu  blinky_dfu.elf  blinky_m0.bin  CMakeCache.txt  cmake_install.cmake  m0.map    Makefile

elal007, 2017-11-14:
It works blinking three LEDs (USB,RX,TX). But when I press reset button, the blinking stop. It means that the program didnot store permanently into RAM.
This is the error when uploading 
elal@ubuntu:~/hackrf/firmware/blinky/build$ dfu-util --device 1fc9:000c --alt 0 --download blinky.dfu 
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: DFU suffix CRC does not match
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Runtime Interface...
Determining device status: state = dfuIDLE, status = 0
dfu-util: WARNING: Runtime device already in DFU state ?!?
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download	[=========================] 100%        11172 bytes
Download done.
dfu-util: unable to read DFU status after completion

I also try using hackrf_spiflash with fail as below
elal@ubuntu:~/hackrf/firmware/blinky/build$ hackrf_spiflash -w blinky.bin 
File size 11140 bytes.
Checking target device compatibility
Compatibility test failed.

dominicgs, 2017-11-14:
That is expected.  If you use DFU mode the firmware image is loaded directly into RAM.

If you want to write it to flash, you can use `hackrf_spiflash -w <firmware_name>.bin`, but beware, that requires firmware support for writing to flash, so writing blinky to your HackRF means you'll need to use DFU mode to recover the original firmware.

elal007, 2017-11-15:
@dominicgs Brilliant.
It was succeeded. Thanks
elal@ubuntu:~/hackrf/firmware/blinky/build$ hackrf_spiflash -i -R -w blinky.bin 
File size 11140 bytes.
Erasing SPI flash.
Writing 11140 bytes at 0x000000.
elal@ubuntu:~/hackrf/firmware/blinky/build$

closed 2017-11-15 by elal007