greatscottgadgets/hackrf #433: hackrf_sweep.exe crashes HackRF upon CTRL+C on Windows

bicurico, 2017-12-04:
### Steps to reproduce
1. Run hackrf_sweep.exe (no arguments needed to reproduce)
2. Stop hackrf_sweep.exe with CTRL+C
3. Run hackrf_sweep.exe (no arguments needed) --> will fail to sweep, unless the RESET button is pressed

### Expected behaviour
It should be possible to terminate hackrf_sweep.exe without crashing the HackRF device.

### Actual behaviour
The HackRF device crashes upon termination of hackrf_sweep.exe with CTRL+C
### Version information
**Operating system**: Windows 10

bicurico, 2017-12-10:
It also crashes HackRF One if hackrf_sweep.exe is run with the -1 option (sweep only one time): after 2-3 calls, HackRF One will crash and RESET button needs to be pressed.

cafeclimber, 2018-03-14:
Also experiencing this issue with the -1 option on Linux

straithe, 2021-03-24:
We've recently done a release which includes some enhances and fixes to `hackrf_sweep`. Can you please try updating to the latest release to see if that fixes your issue?

bicurico, 2021-03-24:
Great!

I will try it out as soon as possible and give some feedback.

straithe, 2021-04-23:
Have you had a chance to see if the latest update addresses the issue you were experiencing?

dmaltsiniotis, 2021-04-23:
> It also crashes HackRF One if hackrf_sweep.exe is run with the -1 option (sweep only one time): after 2-3 calls, HackRF One will crash and RESET button needs to be pressed.

This may be related to #869, if it's the same issue there's a pending PR to address.

I have not had any issues with hackrf_sweep.exe crashing on CTRL+C on windows with the master branch version in cases where the sweep runs for more than 1 second.

straithe, 2021-04-24:
That PR has now been accepted. @bicurico , please let me know if updating from source fixes the issue you are experiencing.

bicurico, 2021-04-24:
Hello,

Sorry for the huge delay in testing this.

I upgraded the FW and used the latest hackrf_sweep.exe contained in the current Pothos distribution and it seems fine.

I was not able to crash it.

Congratulations! And thank you!

Regards,
Vitor

closed 2021-04-24 by bicurico

straithe, 2021-04-24:
Wonderful! Thank you for the update.