greatscottgadgets/hackrf #435: hackrf_stop_rx hangs if a source and a sink are used in gnuradio via gr-osmosdr

gvz, 2017-12-06:
### Steps to reproduce
1. build a flow graph with one "osmocom sink" and one "osmocom source" 
2. modify the generated script in a way that it stops automatically after some time, using tb.stop()
3. start the script and wait

### Expected behaviour
the script should exit after the defined time.

### Actual behaviour
the script waits for the hackrf destructor for eternity. 
After some debugging we found that the usb transfer in "hackrf_set_transceiver_mode" does not return.

### Version information:
git master version of gr-osmosdr and libhackrf
2017.02 hackrf firmware
GNU Radio Companion 3.7.12git-295-ga0adcd33

dominicgs, 2017-12-06:
There are a couple of ways that I'm tempted to tackle this issue:
 * I've already started working on a firmware cleanup branch that changes some of the transceiver mode switching.  This may stop it from failing when the USB call happens. (https://github.com/dominicgs/hackrf/tree/firmware_cleanup)
 * We shouldn't have a timeout of 0 for this call.  Setting a suitably large one before forcibly stopping the HackRF would be a more robust approach.

barto-, 2017-12-07:
Could be related to #394 . As a temporary fix I enabled the LPC43xx watchdog timer

gvz, 2017-12-07:
I think the hackrf does not get stuck. After killing the host programm, the hackrf works normal without resetting.

gvz, 2017-12-07:
ok we found the cause of the hanging usb communication.
It is a problem in osmosdr. Both sink and source are calling hackrf_exit().
It is only call for the last sink or source, that is why it works with multiple sources and sinks.
Is there a way for libhackrf to keep track of opened hackrfs?

PR #437 (2018-02-02): libhackrf: counts the open devices, prevents exit if not all closed
fixes #435, osmosdr calls hackrf_exit for the last sink and the last
source. If both are in one flow graph the desctructor of the source
fails as the sink already closed the usb context.
This prevents hackrf_exit to close the context if not all devices are closed
commit 73ca8c09 (2018-02-02): libhackrf: counts the open devices, prevents exit if not all closed

fixes #435, osmosdr calls hackrf_exit for the last sink and the last
source. If both are in one flow graph the desctructor of the source
fails as the sink already closed the usb context.
This prevents hackrf_exit to close the context if not all devices are closed
closed 2018-02-02 by mossmann