greatscottgadgets/hackrf #436: Windows build artifacts from Appveyor

dominicgs, 2017-12-06:
It would be nice to store the build artifacts from AppVeyor for the adventurous Windows users.

dominicgs, 2018-03-28:
Progress: https://ci.appveyor.com/project/dominicgs/hackrf/build/1.0.101/artifacts

caquino, 2018-04-28:
Is this issue fixed? I'm asking based on your progress as I'm able to see the artifacts available for download, is help still needed?

dominicgs, 2018-04-29:
I haven't tested the binaries and I think I probably need to export the rest of the libraries.  If you'd like to experiment with it, I'd really appreciate the help.

caquino, 2018-05-01:
Sadly looks like appveyor don't let me change the permissions for oauth and it's asking access to all my organizations, which is an issue for me as I can't give access to work-related organizations.

I will try to setup a new github login during the weekend to test this.

closed 2019-03-11 by dominicgs