greatscottgadgets/hackrf #439: Can sample rate changes while sampling fail?

gvz, 2017-12-08:
I am using two hackrfOne between to cots transceivers to emulate a channel between the transceivers.
To emulate all the effect in a real channel I need to change the sample rate of the hackrfs while sampling.
In most cases this is not a problem but sometimes it seems to mess up. In these cases, the cots transceivers are not able to decode the messages sent by the other one.
As this effect does appear at random sample rates far form baseband frequency/2 (Niquist's-theorem), I guess something is going wrong in the hackrf. But debugging this is actually kind of hard.
Are there any ideas. suggestions or experiences with such problems?

dominicgs, 2017-12-08:
Are the two HackRFs changing sample rate at precisely the same time?

Could you share some simple code that does what you're trying to do?

What aspect of a channel are you trying to emulate?

gvz, 2017-12-14:
A college of mine did a "minimal" example with four hackrfs to reproduce the behavior: https://github.com/gvz/hackrf_samplerate_debug
the channel is in hackrf_software_channel.grc and hackrf_software_channel_samp_rate.py uses it and switches the sample rate every second. The sink and source are just have a signal and to see what is happening.

In this example the transmit a 10Hz sine on a 2.48GHz channel, the sine is 1MHz over the center frequency. When the bug tiggers, the sine is 1MHz below the center frequency.
If the bug triggers again the sine is back in its original position.

gvz, 2018-02-16:
by now we found at least one cause of the problem:
At some point while changing the sampling rate the hackRF or the host is messing up the order of I and Q values. This results in the signal is "jumping" to the other side of the center frequency.
We started to clean all buffers in the hackRF and libhackRF while changing the sample rate. That results in the error happening less often but it still happen sometimes.

Another thing we observed during looking to FFTs for quite some time, there are events where the signal does not jump around the center frequency but it is present on both sides with the same energy at the same time. By now we have no clue what happens in this cases.

Any Ideas?

KC5CQW, 2018-03-15:
Quantum Flux. Do it again and see if you get a message back from the USS  Eldridge in 1942 or from Doc Brown in 1985. 

I wonder if there is a way to “hard code” the I Q phase to not shift. Positive vs Negative angles? I’m not a software guy so I’m sure this is rhetorical. In hardware, we use directional couplers, mixers and also passband filters for unwanted sidebands. 

Without knowing more, I’m guessing a shift register... 2’s complement as the inverse of binary missing a conversion back and forth. 

...or a flux capacitor went bad.

closed 2021-02-28 by straithe