greatscottgadgets/hackrf #441: Update HackRF Firmware

DoBa123, 2017-12-11:
### Steps to reproduce
1. https://github.com/mossmann/hackrf/wiki/Updating-Firmware

### Expected behaviour
Update of the HackRF Firmware to the newest Version (2017). 
 
### Actual behaviour
The LEDs were blinking like it's described in the manual. All the steps seemed that they worked out well. But after the Update nothing has changed and the old version is still on the device. 

### Version information
**Operating system**:
I tried a few Linux distributions
Ubuntu
and Gentoo

**hackrf_info output:**
ubuntu@ubuntu:~$ hackrf_info
Found HackRF board 0:
USB descriptor string: 0000000000000000a06063c8221c625f
Board ID Number: 2 (HackRF One)
Firmware Version: 2015.07.2
Part ID Number: 0xa000cb3c 0x00694f5a
Serial Number: 0x00000000 0x00000000 0xa06063c8 0x221c625f

### Output
```
Insert any commandline or build output here
```

dominicgs, 2017-12-11:
Where did you get the binary firmware from?  Did you build it or get it from the release archive?

Have you installed the updated tools too?

DoBa123, 2017-12-12:
I downloaded the firmware from the release archive.
Then I used the Files like it's described in the Chapter Updating Firmware. 
1. updated the SPI
2. updated the CPLD 
What kind of updated tools do you mean?
I don't think I did that. Could you explain that?
Thanks for your respond

miek, 2019-12-02:
Hi, sorry this got overlooked for so long. If you're still having issues with getting your HackRF One updated please re-open this or open a new issue, and we'll get it sorted.

closed 2019-12-02 by miek