greatscottgadgets/hackrf #442: HackRF One not found

abc123me, 2017-12-23:
My HackRF One is not recognized by hackrf_info, I am running a ubuntu chroot on chromeos,whenever attempting to use it I get `hackrf_open() failed: HACKRF_ERROR_NOT_FOUND (-5)` and it is not a usb cable issue as linux recognizes it, here is the result of `dmesg | tail`: 
```dmesg | tail
[ 2779.092956] SELinux: initialized (dev proc, type proc), uses genfs_contexts
[ 2820.909704] usb 1-3: USB disconnect, device number 6
[ 2829.378142] usb 1-3: new high-speed USB device number 7 using xhci_hcd
[ 2829.549949] usb 1-3: New USB device found, idVendor=1d50, idProduct=6089
[ 2829.550002] usb 1-3: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[ 2829.550043] usb 1-3: Product: HackRF One
[ 2829.550073] usb 1-3: Manufacturer: Great Scott Gadgets
[ 2829.550097] usb 1-3: SerialNumber: 0000000000000000a06063c82447765f
[ 2829.553653] platform vpd: Driver vpd requests probe deferral
[ 3048.389901] SELinux: initialized (dev proc, type proc), uses genfs_contexts
```
There is no kernel modules loaded for it (as `lsmod | grep hackrf` returns nothing)

This is not a permissions problem as the rule is installed and I have tried running as root, my RTL-SDR works fine.

dominicgs, 2017-12-23:
This may be a power management issue. I.e. the host is sleeping the device
to save power. We don't support this, so we need to instruct the host to
not do this.

Does running `hackrf_info || hackrf_info` yield different results?

abc123me, 2017-12-24:
No change whatsoever, same exact error

dominicgs, 2017-12-24:
I think this is power mangement getting in our way.

Could you try following the advice in the FAQ here: https://github.com/mossmann/hackrf/wiki/FAQ#power-saving-and-usb-autosuspend

webhat, 2018-01-02:
I'm getting the same error, and I'm not using tlp.

Also blacklisting hackrf in modprobe does not work.

dominicgs, 2018-01-05:
@webhat Are you also running in an Ubuntu chroot on a chromebook?

webhat, 2018-01-08:
Thanks for the help, spoke on IRC it all works now. Loved your #34c3 talk btw.

abc123me, 2018-01-25:
I am going to close this issue since I updated Chrome OS and now its working fine (which indicates that it wasn't my chroot at all but some problem which Chrome OS), sorry for the long response I have been extremely busy with school

closed 2018-01-25 by abc123me

xvostikk, 2019-05-15:
> Thanks for the help, spoke on IRC it all works now. 

@webhat can you describe a solution? we have the same problem..
Thanks in advance!

webhat, 2019-05-16:
> > Thanks for the help, spoke on IRC it all works now.
> 
> @webhat can you describe a solution? we have the same problem..
> Thanks in advance!

As it is so long ago I have difficulty in remembering exactly what the issue was, just that the issues led me here. I think it was that the wrong drivers were being loaded.

According to my zsh history I did the following:
```
rmmod hackrf
apt-get install libhackrf0 libhackf-dev
```
Then I [build this repo from source](https://github.com/mossmann/hackrf/wiki/Operating-System-Tips#building-from-source). Hope this helps.