greatscottgadgets/hackrf #443: HackRF One unable to transmit on Raspberry Pi - HACKRF_ERROR_STREAMING_EXIT_CALLED (-1004)

ansarid, 2018-01-07:
hackrf_transfer seems to stop functioning after about a second when running on a Raspberry Pi. I have not had this issue before since I have used my HackRF One on my Raspberry Pi many times before.

Works fine when recording(-r) but not transmitting(-t).

hackrff_transfer seizes functioning and throws:

> result: HACKRF_ERROR_STREAMING_EXIT_CALLED (-1004)

### Steps to reproduce
1. Install hackrf on (version 2017-11-29 at time of post) on Raspberry Pi 3 Model B+
2. hackrf_transfer -r test.bin -f 303941000 -s 20000000 -l 40
3. hackrf_transfer -t test.bin -f 303941000 -s 20000000 -x 47

> Frequency, gain, and output file makes no difference. I only copied the commands I was using.

### Expected behavior

```
call hackrf_set_sample_rate(20000000 Hz/20.000 MHz)
call hackrf_set_freq(303941000 Hz/303.941 MHz)
Stop with Ctrl-C
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
38.0 MiB / 1.000 sec = 38.0 MiB/second

Exiting... hackrf_is_streaming() result: streaming terminated (-1004)
Total time: 4.00069 s
hackrf_stop_tx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```

> This was run on my PC

### Actual behavior

```
call hackrf_sample_rate_set(20000000 Hz/20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15000000 Hz/15.000 MHz)
call hackrf_set_freq(303941000 Hz/303.941 MHz)
Stop with Ctrl-C
38.3 MiB / 1.000 sec = 38.3 MiB/second
20.2 MiB / 1.000 sec = 20.2 MiB/second

Exiting... hackrf_is_streaming() result: HACKRF_ERROR_STREAMING_EXIT_CALLED (-1004)
Total time: 2.00076 s
hackrf_stop_tx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```

> This was run on my Raspberry Pi

### Version information

2017-11-29-raspbian-stretch
Raspberry Pi 3 Model B+

**hackrf_info output:**

Found HackRF board 0:
USB descriptor string: 0000000000000000a06063c82515465f
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1
Part ID Number: 0xa000cb3c 0x005d4765
Serial Number: 0x00000000 0x00000000 0xa06063c8 0x2515465f

willcode, 2018-01-07:
Take a look at the logs on the RPi (dmesg) to see if the device looks 
like it's disconnecting when you transmit. My guess would be that RPi is 
not supplying enough power for transmit, even though it supplies enough 
for receive.

If you have an externally powered USB hub, give that a try.

Does sample rate make a difference?

ansarid, 2018-01-07:
@willcode dmesg shows the device functioning fine and I have used it without the powered hub before. I tried using it with the powered hub and it still did not work. When I plugged the hackrf back into the Pi the command ran fine. Hmmmmm. I think this is because the HackRF was able to transmit the short signal before it ran into whatever issue. So now I tried recording again, (NOTE: I was trying to record a very short signal, about a second long) but this time I let it record for as long as it could. Here is the output:

```
call hackrf_sample_rate_set(20000000 Hz/20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15000000 Hz/15.000 MHz)
call hackrf_set_freq(303941000 Hz/303.941 MHz)
Stop with Ctrl-C
39.8 MiB / 1.000 sec = 39.8 MiB/second
40.1 MiB / 1.000 sec = 40.1 MiB/second
21.0 MiB / 1.000 sec = 21.0 MiB/second
39.8 MiB / 1.000 sec = 39.8 MiB/second
39.6 MiB / 1.000 sec = 39.6 MiB/second
 0.0 MiB / 1.000 sec =  0.0 MiB/second

Couldn't transfer any bytes for one second.

Exiting... hackrf_is_streaming() result: HACKRF_TRUE (1)
Total time: 6.00170 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit
```
As you can see it was only able to record for about 6 seconds.
Here is the dmesg output from plugging the hackrf in and running the record command:

```
[  745.367102] usb 1-1.2: new high-speed USB device number 18 using dwc_otg
[  745.497698] usb 1-1.2: New USB device found, idVendor=1d50, idProduct=6089
[  745.497712] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=4
[  745.497720] usb 1-1.2: Product: HackRF One
[  745.497728] usb 1-1.2: Manufacturer: Great Scott Gadgets
[  745.497737] usb 1-1.2: SerialNumber: 0000000000000000a06063c82515465f
[  760.234380] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.236559] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.239099] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.241354] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.241563] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.241725] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.241875] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.242004] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.242109] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.242255] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.242415] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.242574] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.242728] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.242873] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.243006] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.243166] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.243324] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.243483] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.243627] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.243759] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.243924] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.244104] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.244257] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.244443] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.244610] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.244751] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.244887] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.245057] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.245224] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.245376] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.245533] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.245694] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.245800] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.245974] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.246131] usb 1-1.2: usbfs: usb_submit_urb returned -121
[  760.246261] usb 1-1.2: usbfs: usb_submit_urb returned -121
[... 1242 lines cut ...]
[ 1640.079693] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.079770] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.079841] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.079912] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.079980] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080050] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080122] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080203] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080277] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080348] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080417] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080509] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080585] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080665] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080748] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080818] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080895] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.080968] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.081039] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.081124] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121
[ 1640.081197] usb 1-1.2.1.4: usbfs: usb_submit_urb returned -121

```

So the results are basically the same.

willcode, 2018-01-08:
I just built the latest hackrf from git on a RPi3, and things are 
working fine with an attached HackRF. I'd suggest that since you are 
using host software from git, that you also update the firmware and cpld 
from git.

https://github.com/mossmann/hackrf/wiki/Updating-Firmware

For reference, my HackRF is running Firmware Version: git-fb83ad6

On 01/07/2018 04:51 PM, Daniyal Ansari wrote:
> @willcode <https://github.com/willcode> dmesg shows the device
> functioning fine and I have used it without the powered hub before. I
> tried using it with the powered hub and it still did not work. [...]

willcode, 2018-01-08:
Hmm, don't bother updating the firmware. I just tried the exact 
configuration (O/S, firmware/cpld and hackrf host software) you have and 
it worked fine. Sorry, I'm not sure what's going on here.

On 01/07/2018 08:54 PM, Jeff Long wrote:
> I just built the latest hackrf from git on a RPi3, and things are
> working fine with an attached HackRF. I'd suggest that since you are
> using host software from git, that you also update the [...]

ansarid, 2018-01-08:
@willcode Can I see your output after transmitting a signal?

willcode, 2018-01-08:
Here is a RX/TX run using /dev/shm. Is it possible that you're running 
into sdcard speed limits?

The HACKRF_ERROR_STREAMING_EXIT_CALLED (-1004) message isn't really an 
error. Hopefully that's not what you were concerned about. In newer 
code, the message was changed to be what you see on your PC.

pi@rpi3:/dev/shm $ hackrf_transfer -r test.bin -f 303941000 -s 20000000 
-l 40

call hackrf_sample_rate_set(20000000 Hz/20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15000000 Hz/15.000 MHz)
call hackrf_set_freq(303941000 Hz/303.941 MHz)
Stop with Ctrl-C
37.5 MiB / 1.000 sec = 37.5 MiB/second
37.5 MiB / 1.000 sec = 37.5 MiB/second
37.5 MiB / 1.000 sec = 37.5 MiB/second
37.5 MiB / 1.000 sec = 37.5 MiB/second
37.5 MiB / 1.001 sec = 37.5 MiB/second
37.5 MiB / 1.001 sec = 37.5 MiB/second
^CCaught signal 2
12.1 MiB / 0.325 sec = 37.1 MiB/second

User cancel, exiting...
Total time: 6.32694 s
hackrf_stop_rx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

pi@rpi3:/dev/shm $ hackrf_transfer -t test.bin -f 303941000 -s 20000000 
-x 47

call hackrf_sample_rate_set(20000000 Hz/20.000 MHz)
call hackrf_baseband_filter_bandwidth_set(15000000 Hz/15.000 MHz)
call hackrf_set_freq(303941000 Hz/303.941 MHz)
Stop with Ctrl-C
38.3 MiB / 1.000 sec = 38.3 MiB/second
38.3 MiB / 1.000 sec = 38.3 MiB/second
38.3 MiB / 1.000 sec = 38.3 MiB/second
38.0 MiB / 1.000 sec = 38.0 MiB/second
38.3 MiB / 1.000 sec = 38.3 MiB/second
38.3 MiB / 1.000 sec = 38.3 MiB/second
  7.9 MiB / 1.000 sec =  7.9 MiB/second

Exiting... hackrf_is_streaming() result: 
HACKRF_ERROR_STREAMING_EXIT_CALLED (-1004)
Total time: 7.00255 s
hackrf_stop_tx() done
hackrf_close() done
hackrf_exit() done
fclose(fd) done
exit

yoyodi, 2018-02-28:
I had the same problem. Thanks to @scateu for helping me solve it. Maybe,  it's also helpful for you to save the recorded file in a RAM disk. 
sudo mkdir -p /ram
sudo mount -t tmpfs -o size=500m tmpfs /ram
hackrf_transfer -t /ram/test.bin -f 303941000 -s 20000000 -x 47

closed 2019-11-27 by miek