greatscottgadgets/hackrf #446: Reverse Bias Diode at Antenna

KC5CQW, 2018-01-16:
Recommendation: add a reverse voltage protection diode to the antenna port that can be switched active/in-line when the antenna bias voltage is not used. Some other protection system may exist beyond a simple diode for RF applications. 

Reason: I applied 12Vdc by accident to the antenna port by an unlabeled bias-T connected in reverse. The HRF was in RX mode with the internal bias off during the incedent but it will no longer TX once the voltage was discovered and removed. “I believe” the excess voltage made a path through the Vaa line (still troubleshooting).

KC5CQW, 2018-01-16:
On my drive home I had a better idea: at the antenna port, add an RF switch that will close the DC path when antenna bias voltage is switched on. Otherwise, the DC path will open and blocked by the input capacitor. This should not affect operation but offer a bit of protection from dum dum’s And 1D 10 T errors. 
This RF switch can be incorporated in a new hardware revision OR an outboard assembly. Actually, it could fit into a custom SMA connector without any additional connections. The antenna bias will throw the switch. 

Other ideas for an outboard idiot box: add a mechanical switch for an internal 50R dummy load for testing the HRF/software setup before normal use, in-line path for normal operations and a short to ground for setup and ESD protection before normal use. This is similar to the “remove before flight” covers on air planes. 
If this isn’t the right forum, please close out and/or delete this issue and commentary.

mossmann, 2022-09-20:
Thanks for the suggestion! We've added a Schottky diode to the DC supply path in HackRF One r9. It is in line both when DC is switched on and when it is switched off. Just in case it is a problem for anyone, we've added a solder jumper bypass.

closed 2022-09-20 by mossmann