greatscottgadgets/hackrf #448: incorrect use of dfu-prefix

mossmann, 2018-01-24:
For some reason we're using dfu-prefix to add a TI Stellaris prefix to our .dfu files.  This is working because it happens to be 8 bytes long and is preceded by the LPC header we subsequently prepend.  Our LPC header is supposed to be 16 bytes long, but ours is only 8 bytes long.  The last 8 bytes get ignored by the bootloader, so the TI Stellaris header is filling that space without any ill effects as far as I know.  Some of those bits are supposed to be high according to the LPC documentation, so I believe the best thing to do is to stop adding the TI Stellaris prefix and to start adding an additional 8 bytes of 0xff onto our LPC header. Replacing those 8 bytes with 0xff worked for me in the one test I've done.

mossmann, 2018-01-24:
see #117

mossmann, 2018-01-24:
There is an LPC header function within dfu-prefix that we could be using instead of our Python solution. I tested it, and it worked.  It does not precisely follow the documentation in UM10503.pdf, however, so I think we should keep using our Python tool.

dominicgs, 2018-03-22:
Fixed in commit 92f122d

closed 2018-03-22 by dominicgs