greatscottgadgets/hackrf #449: DFU boot USB failures

mossmann, 2018-01-24:
### Steps to reproduce
1. Use the DFU bootloader to load hackrf_usb.dfu
2. Repeat until there is an enumeration failure of the firmware running from RAM (It usually fails once within about 50 attempts.)

### Expected behaviour
The firmware should execute from RAM and should enumerate over USB as a HackRF in Hi-Speed USB mode.

### Actual behaviour
Enumeration fails, some messages appear in dmesg indicating failures to enumerate full-speed and/or low-speed devices.  No Hi-Speed device is detected.  Use of an oscilloscope on the USB data lines reveals a never-ending stream of 12 Mbps data produced by the microcontroller.

### Workaround
Reset the device and retry.  Since this bug only happens about 3% of the time, retrying usually works.

### Version information
**Operating system**: observed with various Linux distros

**hackrf_info output:** tested only on 2017.02.1 and more recent firmware

nikoroxy, 2019-11-13:
I've bought a used original board online and i'm only getting this error (DFU and normal boot as well). What can i try to do?

SLAE-Student, 2021-12-06:
looking for any updates on this, I realise this is old, it is now the end of 2021, but I am desperate.  I have a bricked Hack RF One, and it will not get recognised by anything. I have tried multiple Linux dists and flavours, (Debian, Kali, Redhat, Pentoo, etc and Win 8,10, Mac OSX). I have tried every hint I can find on the internet and nothing seems to work. Is there a way to bypass USB and reset using GPIO? I will continue to check here and GSGadgets for any changes or updates to this issue wherever it appears on the internet Thanks.

gozu42, 2021-12-06:
> I have a bricked Hack RF One, and it will not get recognised by anything.

that seems unlikely to be related to this particular issue.

please open a separate issue for your problem, and given your description, include results (dmesg) of an attempted DFU boot (press and hold DFU button, press and release reset button, release dfu button) and pictures of the front and back of the hackrf board.

martinling, 2022-06-13:
Possibly related to the PLL issues suspected to be behind #1101. Retest with #1106.