greatscottgadgets/hackrf #450: hackrf_sample_rate_set() error

archon247, 2018-01-25:
### Steps to reproduce
1. plug hackrf into usb slot
2. install driver unsing zadig
3. install SDR#
4. configure SDR# to use the hackrf
5. press play to scan

### Expected behaviour
Should start listening

### Actual behaviour
hackrf_sample_rate_set() error

**Operating system**: Windows 10 pro x64

is there something im doing wrong setting up the hackrf

miek, 2019-11-27:
Sorry this never got a response originally. I'm going to close this assuming this was solved, as it's really old, but if you are still having this issue please re-open or open a new issue!

closed 2019-11-27 by miek

yo-wotop, 2021-04-09:
I just ran into this error out of nowhere when trying to use SDR#. I then tried opening HDSDR and got a different error (No HackRF device detected). Unplugging/replugging the HackRF USB solved this issue for me (for both pieces of software).