greatscottgadgets/hackrf #451: Time and Clock Synchronisation

elal007, 2018-01-25:
### Steps to reproduce
1. Two HackRFs was used. One as transmitter (TX), the other one as Receiver (RX)
2. The Transmitter will transmit Beacon cw.
3. In the receiver, the signal received will auto-correlate. The binary sequence with good autocorrelation properties is used.

I have a problem. I need a suggestion. I want to use ranging based on Time of Arrival. In terms of the sampling rate maximum with hackrf is 20Msps. I just curious how can I synchronize the clock and time of two HackRF devices. 
1. Is possible to synchronize the devices with software-based (not using external clock h/w).
2. If the external clock is needed, please suggest me how to do it carefully. I am afraid, it will damage the device itself.

Also, I found the article here: https://github.com/mossmann/hackrf/wiki/Multiple-device-hardware-level-syncronisation. But, that kind of synchronisation is not used for ranging purpose. Am I right?

or the paper was written by Bartolucci on http://spcomnav.uab.es/docs/conferences/Bartolucci_NAVITEC_2016.pdf

Cheers,

barto-, 2018-01-25:
Hi,
Your best option at the moment is #381 . I had good results with tdoa localization using this technique. As long as your observation time is short enough (in the order of few ms), you should be able to get good results without synchronizing the clock. However, you need to provide time synchronization using the SYNC_IN pin. TOA is a bit more complex and you may need to synchronize the clock.

elal007, 2018-01-28:
@barto- Thank you for the suggestion. I will study your suggestion.

KC5CQW, 2018-03-15:
If the two HRF’s are close, use coax between the clock out of one and clock in of the other. You must account for the velocity factor of the coax at 10MHz and calculate the group delay between them. 

Otherwise if they are some distance apart, a GPSDO on each HRF clock input. The expected in and out clock signal is a 3.3Vdc square wave. You will need HCMOS type GPSDO output. I would recommend this method over a simple coax link. 
Most GPSDO’s are capable of 10^-9 to 10^-12 precision. 

I have a surplus TruePosition GPSDO and a modified video distribution amplifier (50ohm terminations, it comes as 75 ohm for video service) from eBay. 

TruePosition units were used for E911 location calculations at cell towers. With all towers synced to GPS, they could triangulate the position of a cell phone. The accuracy was limited to the DOP Dilution of Precision at each tower. However,... smart guys have found several ways to factor this variable out. 
In any case, some inaccuracy is expected from a transmitter due to multi-path propagation as the signal is reflected from nearby structures. This would/could be factored out further provided the transmitter (cell phone) has a reasonable clock sync as well. Cell towers provide this time as derived from their GPSDO and calculate the propagation delays.

I would recommend you use a GPSDO.

elal007, 2018-03-20:
Thank @KC5CQW ,
My plan is to use the TOA indoor, so I will not use GPS. I will try to use the Coax cable. But now, I have another problem with the HackRF. I will raise the issue later in this forum. Thanks for your suggestion. Really appreciate it.

straithe, 2021-01-28:
Are you still experiencing this issue or can this ticket now be closed?

closed 2021-02-28 by straithe