greatscottgadgets/hackrf #454: x2 Hack RF One SDRs will not connect to GQRX / GRC (hackrf_info DOES work, firmware update successful)

mrroboto1420, 2018-02-08:
### Steps to reproduce
1.  Install GNU Radio environment using:
`sudo apt-get install gqrx-sdr`
2. Updated firmware each Hack RF One
3. Try to open GQRX or a GRC Example with one SDR connected

### Expected behaviour
GQRX / GRC Example detects the Hack RF One and auto populates the device string.  Everything works fine.
### Actual behaviour
GQRX / GRC Example does not detect the Hack RF One.  Note that this happens with two Hack RF Ones, both updated firmware no problem but neither can connect to GQRX / GRC Example.
### Version information
**Operating system**:
KDE Plasma Version:  5.10.5
KDE Frameworks Version: 5.38.0
Qt Version: 5.9.1
Kernel Version: 4.13.0-31-generic
OS Type: 64-bit

**hackrf_info output:**
`hackrf_info version: unknown
libhackrf version: unknown (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c8255d655f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x005e475e
`

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

GNU Radio Companion 3.7.10
Gqrx 2.10

### Output
```
Insert any commandline or build output here
```

mossmann, 2018-02-13:
Can you try configuring the device string in gqrx to "hackrf=0"?

dominicgs, 2018-04-01:
@mrroboto1420 Have you had chance to try specifying the device using "hackrf=0" as @mossmann suggested?

mrroboto1420, 2018-04-02:
I think it was something with my OS and the library installs.  I was using
a fresh Kubuntu 17.10 install.  I just installed regular Ubuntu and
followed the steps again and everything works fine.

dominicgs, 2018-04-02:
Thanks great, thanks for the update.

I'm going to close this issue now.

closed 2018-04-02 by dominicgs