greatscottgadgets/hackrf #455: HackRF no RX / TX (one more time)

detjar, 2018-02-08:
I bought HackRF one. Yeah :-)

The current result of hackrf_info is as below. Firmware version was 2015.xxxx but I decided to upgrade it, as I did not know what is happening (try & fails method).

`hackrf_info version: git-991c508
libhackrf version: git-991c508 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a27466e626520b0f
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00694758`

My issue was almost as this one #https://github.com/mossmann/hackrf/issues/242 except that I have couple of antennas and none is getting any signal @ SDR#. 

Let's say I want to find some FM station - nor antenna, neither paper clip is working for me, however the RX LED is on - no signal at SDR# (WFM).

I tried to transmitt something - TX LED is on then, but no signal is produced over the antenna.

Looks like something is broken (since I receive it I guess?). Can anyone help me debugging / diagnosing the core issue over there?

https://imgur.com/a/nqwLu

unixpunk, 2018-02-08:
Can you send screenshot of the settings window for the device?  (The gear
button up top) SDR# hides the gain/amp settings in there.  First, make
sure the LNA/amp is off to start with and then set the gains both at 32 or
so.  Does that help?

--Steve

I bought HackRF one. Yeah :-)

The current result of hackrf_info is as below. Firmware version was
2015.xxxx but I decided to upgrade it, as I did not know what is happening
(try & fails method).

`hackrf_info version: git-991c508
libhackrf version: git-991c508 (0.5)
Found HackRF
Index: 0
Serial number: 0000000000000000a27466e626520b0f
Board ID Number: 2 (HackRF One)
Firmware Version: 2017.02.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00694758`

My issue was almost as this one
#https://github.com/mossmann/hackrf/issues/242 except that I have couple
of antennas and none is getting any signal @ SDR#.

Let's say I want to find some FM station - nor antenna, neither paper clip
is working for me, however the RX LED is on - no signal at SDR# (WFM).

I tried to transmitt something - TX LED is on then, but no signal is
produced over the antenna.

Looks like something is broken (since I receive it I guess?). Can anyone
help me debugging / diagnosing the core issue over there?

https://imgur.com/a/nqwLu

detjar, 2018-02-08:
https://imgur.com/obQDoSW

LNA/amp was off. It does not help at all if I change LNA/amp.

unixpunk, 2018-02-08:
Your gains are all down.  No gain means no signal.  Turn both to 32 and
see what you get, probably too much gain, but you need to figure that part
out on your own.

--Steve

https://imgur.com/obQDoSW

LNA/amp was off. It does not help at all if I change LNA/amp.

detjar, 2018-02-08:
k, thanks. Will do so.

detjar, 2018-02-10:
@unixpunk https://youtu.be/LFactIUsd08 that is the result (tested on my laptop this time). Will take a look with pentoo distro as well, but I don't expect any difference. Any ideas?

unixpunk, 2018-02-11:
@detjar try reducing the sample rate down to 2 first, waterfall looks
choppy.  Then might need a better antenna, seen issues like this with
lower quality antennas, might even try touching it with your hand too. 
Best bet, plug your stereo receiver antenna in to HackRF.  :)  HTH

unixpunk, 2018-02-11:
Even the ANT500 is not that great for Rx, so I can't imaging its really
good for Tx either but should work if bandwidth isn't too large I think. 
Glad its just the antenna though!

mossmann, 2018-02-13:
Thanks for helping, @unixpunk!

closed 2018-02-13 by mossmann

unixpunk, 2018-02-13:
YW, I do it for a living @ Liquidware.  :)