greatscottgadgets/hackrf #456: Hack RF not detecting, firmware update not working

SujayNarayana, 2018-02-10:
### Steps to reproduce
1. Connect HackRF to PC

### Expected behaviour
HackRF should be detected in PC. USB LED should glow

### Actual behaviour
HackRF is detected as Unrecognized device. Only 3V, 1.8V and RF LEDs are glowing.

### Version information
**Operating system**:
Win 7

**DESCRIPTION:**
I connected HackRF to PC. It is being recognized as "Unknown device". I had installed the HackRF driver from Zadig earlier and it was working. Now, all of a sudden it is not working. Hackrf_tools is not detecting HackRF. 

I tried updating dfu firmware as mentioned in the webpage. In DFU mode, Only RF and 3V LEDS glow.
The output is as follows.:

**hackrf_info output:**

_dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        22056 bytes
Download done.
unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode_

After this the hackrf disconnects from PC and as earlier 3.3V, 1.8V and RF LEDS glow.

I tried programming LPC4320 with the latest firmware "hackrf_one_usb.bin" using JTAG. I could successfully program HackRF but no luck, same behavior.

PCB has no burn mark or burn smell.

mossmann, 2018-02-13:
Can you please try connecting the HackRF to Linux PC?  If you do not have a Linux installation, I suggest downloading the [GNU Radio Live SDR Environment](https://wiki.gnuradio.org/index.php/GNU_Radio_Live_SDR_Environment)

mossmann, 2018-02-13:
Once connected to a Linux box, check to see if the USB LED lights up and look for 1d50 in the output of `lsusb`.

SujayNarayana, 2018-02-13:
Tried that earlier. USB LED doesnt glow even in linux PC. Also, lsusb doesnt show HackRF. However, DFU mode works in linux as well as windows. It is detected ass LPC.

mossmann, 2018-02-20:
Which release did you take your hackrf_one_usb.bin from?

Please try this:
1. Boot in DFU mode
2. dfu-util --device 1fc9:000c --alt 0 --download hackrf_usb.dfu
3. Check for 1d50 in lsusb.
4. Assuming 1d50 has failed to appear, check the status of the RF LED and report it here.  Is it off, dimly lit, or fully lit?
5. Hold down the DFU button while pressing and releasing the RESET button. (I want you to bring it back into DFU mode without removing power.)
6. Check for NXP in lsusb and also check the status of the RF LED again and report these results here.
7. retry: dfu-util --device 1fc9:000c --alt 0 --download hackrf_usb.dfu
8. Does 1d50 appear in lsusb this time?

SujayNarayana, 2018-02-21:
Which release did you take your hackrf_one_usb.bin from?
**--> From the github release page. The release version is release 2018.01.1**

Please try this:

    Boot in DFU mode
    dfu-util --device 1fc9:000c --alt 0 --download hackrf_usb.dfu
**--> Used "hackrf_one_usb.dfu". In dfu mode, 3V3 and RF LED glows (full bright). Once dfu update is complete, 3V3, 1V8 and RF LEDs glow brightly.** 
    Check for 1d50 in lsusb.
**--> Does not appear. If I try the same in windows, HackRF is detected as unrecognized device (same as when when we connect broken USB devices)**
    Assuming 1d50 has failed to appear, check the status of the RF LED and report it here. Is it off, dimly lit, or fully lit?
**--> Fully lit**
    Hold down the DFU button while pressing and releasing the RESET button. (I want you to bring it back into DFU mode without removing power.)
    Check for NXP in lsusb and also check the status of the RF LED again and report these results here.
    retry: dfu-util --device 1fc9:000c --alt 0 --download hackrf_usb.dfu
    Does 1d50 appear in lsusb this time?

**--> Same behavior as before**

mossmann, 2018-02-23:
Can you try using this [test firmware](https://greatscottgadgets.com/hackrf-test.zip)?

SujayNarayana, 2018-02-24:
@mossmann: Thank you very much for your support. Here are the observations.

JTAG is the only possible way for me to program HackRF. I programmed it with blinky.bin. All the LEDs are blinking as expected.

With test firmware no improvement. Same observation.

I did not go through the hackrf source code but just curious if HackRF waits for some reply from other devices such as FPGA before initializing USB wherein the other device went bad?

More observations: 

1. Hold DFU button -> Connect USB to PC -> Release DFU button -----> 3V3 LED glows brightly, TX LED at 50% brightness, 1V8 LED OFF, rest of the LEDs are at around 10% brightness.

2. With HackRF connected to PC, hold DFU button -> press and release reset button -----> 3V3 LED glows brightly, RF LED at full brightness, TX at 50%, 1V8 LED OFF, rest are at around 10% brightness.

SujayNarayana, 2018-02-24:
I also tried to load the test firmware using JTAG. Even though programming was successful, USB LED is not glowing. Same observation as before.

f6jig, 2018-04-05:
i have thé same problem.
m'y hacker one is only detected on dfu mode. my USB led lever illuminated outside this mode.
i dont understand could you tell me toi

zanco, 2018-07-29:
Hi, (Chinese HackRF version) Might or might not be helpfull for anyone, but after disconnecting my HackRF from the main computer and connect it to the laptop my HacRF did not light the USB led anymore and other led's not as bright. Tried other micro USB cable and computer with no luck. 

Opened the case, powered the unit and noticed 2 Volt drop over FB1 which is power to the unit. On the press of the pin of my multimeter the USB led lighted up.

So I resoldered FB1 and all works fine now!

(My first original I blew because connected to a non earthed PC and earthed antenna system. So my PC got ground by USB shield and antenna cable. Even the processor USB connection shorted to ground, game over it was...)

Ferenczyg, 2018-08-02:
Hi Zanco i have exactly the same problem. Could you please give me more info about FB1? How do I locate it in the mainboard?

ekiojp, 2018-12-03:
Just in case anyone face this behavior, after trying **8 different micro-usb cables**, last one work!
Clearly a poor quality cable will make the USB LED not coming up and neither `lsusb` will see it
[README](https://github.com/mossmann/hackrf/wiki/Tips-and-Tricks)

SujayNarayana, 2018-12-03:
I used JTAG to execute example programs on the microcontroller. All the programs including LED blinking worked except USB. Looked like the USB PHY side of the MCU is damaged. Replaced the microcontroller now and Hack RF is working fine :)

closed 2018-12-03 by SujayNarayana

novaKC3, 2019-06-22:
This happened to me.  Here's what worked.  Leave powered off for about an hour (maybe because some electronics retain power even after disconnected from main power source).  The, plug into you PC (mine is Windows 10). Then hold down the DFU and reset buttons for about five seconds together.  Then all my lights came on and it was connected.

Commander-Crash, 2021-03-14:
@SujayNarayana I know this is a closed thread however I'm having this issue it came out of nowhere. I have hackrf one chinese clone Portapack H2 firmware. Windows detects it but no usb light no software detects device windows makes the sound as if usb is connect but do not see any change in device manager. lsusb in linux will not recognize it. Used over 20 micro usb cables 5 of them i bought new. No go. Has someone found a fix that does not require soldering a replacement MCU or purchasing a hackrf again?

ka9cql, 2021-03-27:
@SujayNarayana I, too have a Chinese clone that got messed up when I tried to upgrade from the Portapack it had on it from the manufacturer to the Mayhem Portapack H2 release. I haven't managed to restore the factory firmware yet (I suspect it may be customized) but I can use the following process to get my device working under Linux as a HackRF One.

 1400  Plug HackRF One (HRF) into Linux computer USB port
 1410  After power-up, reset it by pressing the reset button
 1411  Using lsusb I do not see NXP device
 1413  Press and hold both the Reset button and the DFU button, wait a second or two, release Reset, pause, then release DFU button
 1414  Using lsusb I can now see an NXP Semiconductors device (Great!)
 1416   dfu-util --device 1fc9:000c --alt 0 --download hackrf_one_usb.dfu
 1417  Seems to download the dfu, but errors out after writing (device not found, I believe)
 1418   hackrf_info now shows the HRF device! (More goodness!)
 1419  Loading my SDR porgram (CubicSDR is what I tried) now seems to work.  It detects my unit as a regular Hack RF One.

I will keep at it, and update this thread with more information.

My goals are to figure out the Windows equivalent of the above procedure to be able to use my Hack RF One on Windows.

A stretch goal is to (obviously) fully restore the device to its its state when I opened the box.

A final goal of mine is to push the Mayhem H2 PortaPack onto the device in place of what it shipped with.

ka9cql, 2021-03-27:
WINDOWS -

Download Zadiag.exe
Power up the HRF
Press and hold both the Reset button and the DFU button
Release ONLY the Reset button
wait a second or two
Release the DFU button
Check device manager. I found an "LPD device" but it said it wasn't working
Launch Zadiag
Select Options/List All Devices
In the device drop-down list, select LPD device
Used Zadiag to replace whatever driver Windows is currently using with libusbK (v3.0.7.0)
Now my DFU utility can find the device! (good!)

C:>dfu-util-static.exe -l
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Found DFU: [1fc9:000c] ver=0100, devnum=25, cfg=1, intf=0, path="1-8", alt=0, name="DFU", serial="ABCD"

Received a close-command error.

Use the dfu-util-static.exe utility to download the HackRF One dfu -

C:>dfu-util-static.exe --download hackrf_one_usb.dfu
dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        22056 bytes
Download done.
unable to read DFU status after completion

Then flash the device with the HackRF One firmware -

C:>dfu_hackrf_one.bat
*** Run HackRF firmware in RAM via LPC DFU ***

This is used to "unbrick" your HackRF, if you are no longer able to use
HackRF tools to flash or operate your HackRF.

Connect your HackRF One to a USB port on your computer.

Hold down both the DFU and RESET buttons on the HackRF.
Then release the RESET button (closest to the edge).
Then release the DFU button.

Press any key to continue . . .

dfu-util 0.9

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2016 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

DFU suffix CRC does not match
A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 1fc9:000c
Run-time device DFU version 0100
Claiming USB DFU Interface...
Setting Alternate Setting #0 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0100
Device returned transfer size 2048
Copying data from PC to DFU device
Download        [=========================] 100%        22056 bytes
Download done.
unable to read DFU status after completion
can't detach
Resetting USB to switch back to runtime mode

Press any key to continue . . .

C:>

Device now shows up as a HackRF One!
Device Manager sees it as a USB device called "HackRF One"!
Zadiag says that the "HackRF One" device is using the WinUSB (v6.1.7600.16385) driver (I have overruled the v.10 one, earlier) BUT YOU PROBABLY DON'T WANT TO CHANGE WHATEVER THIS IS ON YOUR BOX!

Just leave it be, and see if Device Manager can find a HackRF One under the "USB Serial Device" section

Now test it as a Windows SDR.

I ran AirSPY (was sdrsharp.exe) and IT WORKS!

So this is how I got it working as an SDR under Windows.

Still can't boot it up, stand-alone, and it's only working as an SDR after I follow this procedure every time I plug it in.  If you remove the USB cable, it doesn't work.

At least I'm making progress!

ka9cql, 2021-04-12:
I finally got my Chinese PortaPack H2 clone to work.

I used the procedures above to restore my device to "native HackRF mode" (aka disconnect built-in battery, and follow the above procedure to restore "native" HackRF mode), and then downloaded the PortaPack H2 firmware from the following URL -

http://www.mmdvm.club/index.php/h2.html

As of today (2021-04-12) file/version mayhem_v1.3.1_FIRMWARE_WM8731.zip was present.

After downloading that zip onto a Windows 10 box, I unzipped it and changed directory to the newly-unzipped directory.

I plugged my HackRF into my Windows 10 box and then ran "flash_portapack_mayhem_8731.bat".  VIOLA!  PortaPack H2 operation restored!    W0000 H0000!

Here are the contents of flash_portapack_mayhem.8731bat -
@echo off

echo *** Re-flash the HackRF with PortaPack wm8731 firmware from www.mmdvm.club***
echo.
echo Connect your HackRF One to a USB port on your computer.
echo.
echo If using a PortaPack, put the PortaPack in HackRF mode by selecting
echo the "HackRF" option from the main menu.
echo.
pause

echo.
hackrf_update.exe portapack-h1_h2-mayhem.bin
echo.
pause

     SIZE        FILENAME
---------------------------------------------------------
    865,620 portapack-h1_h2-mayhem.bin

I have attached the portapack-h1_h2-mayhem.bin I used to this post

Enjoy!
[ChineseCloneOf__mayhem_v1.3.1_FIRMWARE_WM8731.zip](https://github.com/mossmann/hackrf/files/6300177/ChineseCloneOf__mayhem_v1.3.1_FIRMWARE_WM8731.zip)

EticalHack, 2023-01-12:
Hi u all! Om new in here but i have search and search on the web all over but cant find whats housing this problem. I at first bought hackrf with portapack2 and then the buttons were done and the screen was crusched.....so i ordered i new portapack2

But for months now i cant solve what causing the issue for hackrf Alone... Somebody have messed with me cops or somebody else but i didnt recognize my hackrf pcb anymore....can someone tells me what the problem is? Should the pcb looken like this or am i stupid?

I think the pcb lack a component or two and the also have been soldering off the board.

 

The only thing s whole computers can tell me is that the hackrf is found as LPC nothing else could be done. No upgrade no downgrade or flashing..... And the leds are they even supposed to look like this im gonna show u with a pic..

I just plugged in the device in my linux laptop and wrote lsusb and the information i got from the usb ports was this>

Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 04f2:b5fa Chicony Electronics Co., Ltd EasyCamera
Bus 001 Device 002: ID 8087:0a2a Intel Corp. Bluetooth wireless interface
Bus 001 Device 006: ID 1fc9:000c NXP Semiconductors LPC4330FET180 [ARM Cortex M4 + M0] (device firmware upgrade mode)

Please can someone telll me what to do next, in windows i get the hackrf detected as LPC nothing more.....

Can i do something about it in ubuntu to upgrade something, i dont know what to do next, but in linux it is detecting the hackrf as NXP Semiconductors LPC4330FET180 [ARM Cortex M4 + M0] (device firmware upgrade mode)

but i dont know what do do or next step.. can somebody please help me out! thanks alot

/Eticalhack

i could upload a photo or two of the pcb incase somebody want it....