greatscottgadgets/hackrf #457: Install API in Windows

pachoduran, 2018-02-15:
Hi, need  your help, need install HackRf API en Windows , need work in c#. Please help me.

Thanks

pachoduran, 2018-02-17:
Hi, help me please, i can´t install if Windows libhackrf using information in:

https://github.com/mossmann/hackrf/tree/master/host

Please help me

James-Brown, 2018-02-17:
I was successful building HackRF host software using Visual Studio 2015. This might help
http://www.seti.net/notebook/volume-10/Chap66/chap66.php
Regards......... Jim

From: pachoduran 
Sent: Saturday, February 17, 2018 05:09
To: mossmann/hackrf 
Cc: Subscribed 
Subject: Re: [mossmann/hackrf] Install API in Windows (#457)

Hi, help me please, i can´t install if Windows libhackrf using information in:

https://github.com/mossmann/hackrf/tree/master/host

Please help me

pachoduran, 2018-02-17:
Hi . Yesssssssssssssss

Work very fine all information. I can do install.  Thanks, Thanks, Thanks

I use the release 2017.02.1 and funtion very good.

krh3o, 2018-02-18:
Thank you Jim!! I spent hours just trying to flash the hackrf. The wiki instructions for Windows are close to useless, but your steps got the tools built and I was able to flash my device.

James-Brown, 2018-02-18:
My pleasure Karl.

From: Karl Waters 
Sent: Saturday, February 17, 2018 22:04
To: mossmann/hackrf 
Cc: James Brown ; Comment 
Subject: Re: [mossmann/hackrf] Install API in Windows (#457)

Thank you Jim!! I spent hours just trying to flash the hackrf. The wiki instructions for Windows are close to useless, but your steps got the tools built and I was able to flash my device.

pachoduran, 2018-02-18:
Hi. Thanks for all your help jim. 

Another question, your can send to me samples of how work in c# or VB with hackrf api???

thanks

Bye

pachoduran, 2018-02-20:
Hi, Sorry for me english. 

Hi, I could not import the hackrf.dll library in c # or vb. errors always come out. Somebody could help me.

Thank you very much

zillant, 2019-04-06:
Hello!
I can't install API, as showned in this tutorial -> http://www.seti.net/notebook/volume-10/Chap66/chap66.php

> CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
> Please set them or make sure they are set and tested correctly in the CMake files:
> FFTW_INCLUDES
>    used as include directory in directory C:/hackrf/host/hackrf-tools/src

Please, help me. This is very important for my project with hackfr tools.

James-Brown, 2019-04-06:
Zillanat – What step failed for you.
Contact me directly and we can work it out. I may have an error on my web site.
Jim@SETI.Net

From: zillant 
Sent: Saturday, April 6, 2019 00:37
To: mossmann/hackrf 
Cc: James Brown ; Comment 
Subject: Re: [mossmann/hackrf] Install API in Windows (#457)

Hello!
I can't install API, as showned in this tutorial -> http://www.seti.net/notebook/volume-10/Chap66/chap66.php

  CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
  Please set them or make sure they are set and tested correctly in the CMake files:
  FFTW_INCLUDES
  used as include directory in directory C:/hackrf/host/hackrf-tools/src

Please, help me. This is very important for my project with hackfr tools.

garik7777, 2019-04-06:
Good day,thank you for responding and help me when it's convenient for you to start ?  how will we communicate ? I have Skype, viber,  https://discordapp.com  my discordapp MOZG#9284 but if you have  https://www.teamviewer.com/ru/  it will be much easier my teamviewer 102227063,in advance thank you very much, I will be grateful,today I will wait when It will be convenient for you ?

>
>Zillanat – What step failed for you.
>Contact me directly and we can work it out. I may have an error on my web site.
>Jim@SETI.Net
>
>From: zillant 
>Sent: Saturday, April 6, 2019 00:37
>To: mossmann/hackrf 
>Cc: James Brown ; Comment 
>
>Hello!
>I can't install API, as showned in this tutorial ->  http://www.seti.net/notebook/volume-10/Chap66/chap66.php
>
>CMake Error: The following variables are used in this project, but they are set to NOTFOUND.
>Please set them or make sure they are set and tested correctly in the CMake files:
>FFTW_INCLUDES
>used as include directory in directory C:/hackrf/host/hackrf-tools/src
>
>Please, help me. This is very important for my project with hackfr tools.

zillant, 2019-04-07:
> Zillanat – What step failed for you.
It's 12 step. I copied and pasted to command line. I think, this error occurs, because this tutorial for older version of hackrf_tool.

James-Brown, 2019-04-07:
I’ll go over it again.
I have moved to Visual Studio 2019 now. Which version do you use?

From: zillant 
Sent: Sunday, April 7, 2019 00:45
To: mossmann/hackrf 
Cc: James Brown ; Comment 
Subject: Re: [mossmann/hackrf] Install API in Windows (#457)

  Zillanat – What step failed for you.
  It's 12 step. I copied and pasted to command line. I think, this error occurs, because this tutorial for older version of hackrf_tool.

zillant, 2019-04-07:
At first try i used 2017, after that i used 2015 version.
> I’ll go over it again.
I have moved to Visual Studio 2019 now. Which version do you use?

James-Brown, 2019-04-08:
I just switched to 2019

From: zillant 
Sent: Sunday, April 7, 2019 10:02
To: mossmann/hackrf 
Cc: James Brown ; Comment 
Subject: Re: [mossmann/hackrf] Install API in Windows (#457)

At first try i used 2017, after that i used 2015 version.

  I’ll go over it again.
  I have moved to Visual Studio 2019 now. Which version do you use?

James-Brown, 2019-04-09:
I am sorry – I cannot get the build process to work with VS 2019. I am not knowledgeable enough to debug it. 
If someone else can help it would be appreciated.

Regards..... 

From: zillant 
Sent: Sunday, April 7, 2019 10:02
To: mossmann/hackrf 
Cc: James Brown ; Comment 
Subject: Re: [mossmann/hackrf] Install API in Windows (#457)

At first try i used 2017, after that i used 2015 version.

  I’ll go over it again.
  I have moved to Visual Studio 2019 now. Which version do you use?

twchia, 2019-08-12:
I am trying to guild the hackrf.dll for c# reference use as indicated by the link 
 http://www.seti.net/notebook/volume-10/Chap66/chap66.php

At the last stage of the cmake I get the error that the library file cannot be found when the pthreadVC2.lib file is present
[image]

Can anyone advice the possible issue that I am encountering.

Hoernchen, 2019-08-12:
The backslash at the end of the line means the line continues in the next row (with a newline!), but you're pasting one big blob. If you want to paste it as one long line, remove all these \ at the end of all rows.
Additionally, the proper continuation character for windows is ^ - try pasting all of this at once:
```
cmake ../ -G "Visual Studio 16 2019" ^
-DLIBUSB_INCLUDE_DIR=C:\libusb_master\x64\Release\lib\libusb-1.0 (static) ^
-DLIBUSB_LIBRARIES=C:\libusb_master\x64\Release\lib\libusb-1.0.lib ^
-DTHREADS_PTHREADS_INCLUDE_DIR=c:\pthreads-w32-2-9-1-release\Prebuilt.2\include ^
-DTHREADS_PTHREADS_WIN32_LIBRARY=c:\pthreads-w32-2-9-1-release\Prebuilt.2\lib\x64\pthreadVC2.lib
```

Hoernchen, 2019-08-12:
Oh, and there is a random unescaped space in your path that fucks everything up: libusb-1.0 (static)
I have honestly no idea who made up that line on that website, with so many errors...

James-Brown, 2019-08-12:
That would be me and my web site.
I fixed the web site per your suggestions but I have not yet retested it.
That area of the instructions was marked as not working but I hope it is now.
Any retest would be appreciated.
Regards........ James Brown W6KYP

From: Hoernchen 
Sent: Monday, August 12, 2019 09:27
To: mossmann/hackrf 
Cc: James Brown ; Comment 
Subject: Re: [mossmann/hackrf] Install API in Windows (#457)

Oh, and there is a random unescaped space in your path that fucks everything up: libusb-1.0 (static)
I have honestly no idea who made up that line on that website, with so many errors...

twchia, 2019-08-13:
> ```
> cmake ../ -G "Visual Studio 16 2019" ^
> -DLIBUSB_INCLUDE_DIR=C:\libusb_master\x64\Release\lib\libusb-1.0 ^
> -DLIBUSB_LIBRARIES=C:\libusb_master\x64\Release\lib\libusb-1.0.lib ^
> -DTHREADS_PTHREADS_INCLUDE_DIR=c:\pthreads-w32-2-9-1-release\Prebuilt.2\include ^
> -DTHREADS_PTHREADS_WIN32_LIBRARY=c:\pthreads-w32-2-9-1-release\Prebuilt.2\lib\x64\pthreadVC2.lib
> ```

Thank you. However, I still encounter the following issue.

[image]

twchia, 2019-08-13:
Can anyone show me a VS2019 c# solution that has references to hackrf api and a basic usage such as how to use the api to initiate a connection to the SDR via USB interface?
This will save the hassle and instead allow me to work on my project directly,

Thank you.

Hoernchen, 2019-08-13:
The build instructions provided by the somewhat dusty readme at https://github.com/mossmann/hackrf/tree/master/host include handling of fftw3

sreverol, 2021-07-15:
> Hi, need your help, need install HackRf API en Windows , need work in c#. Please help me.
> 
> Thanks

Hi, did yoy find how to accomplish this?

Thanks!

maitbot, 2025-08-14:
I've refactored the CMake build system over at
https://github.com/maitbot/hackrf/tree/mait/update-cmake

Please test if you can.

The thing that keeps me from making a pull request is that I need help ensuring that it does not break things for those using a Windows build.

closed 2026-01-04 by mossmann