greatscottgadgets/hackrf #458: Signal at 2x freq offset with v2018.01.1-172-g991c508, rel 2018.01.1 works as expected.

donmr, 2018-02-17:
### Steps to reproduce
1.  HackRF tools and FW from git:    v2018.01.1-172-g991c508
2.  CW signal at 7.250 MHz, very narrow.
3.  Sample with:       hackrf_transfer -f 7200000 -s 400000 -n 400000 -w
     OR run osmocom_fft with same settings

### Expected behaviour
FFT of should shows signal 7.250 MHz

### Actual behaviour
FFT of shows signal at 7.300 MHz!

*** Using Release 2018.01.1 works as expected ***

### Version information
Mint 18.1 (Ubuntu 16.04 based)

Receive wav file: HackRF_20180217_103851Z_7200kHz_IQ.wav
call hackrf_set_sample_rate(400000 Hz/0.400 MHz)
call hackrf_set_hw_sync_mode(0)
call hackrf_set_freq(7200000 Hz/7.200 MHz)
samples_to_xfer 400000/0Mio
Stop with Ctrl-C
 0.3 MiB / 1.000 sec =  0.3 MiB/second
 ...

I'll be happy to test this at other frequencies etc if that will help.

mossmann, 2018-02-17:
Did you update the firmware without updating the CPLD?

donmr, 2018-02-17:
Yes, will fix that.

donmr, 2018-02-18:
That did it, thanks.  May I suggest that a mention of the cpld code be added to the firmware/README file?

(Sorry for the delay, I had an unplanned system rebuild interrupt things)

mossmann, 2018-02-20:
I'm reopening this to serve as a reminder to add a note about CPLD update to the release notes for the next release.

jboone, 2018-02-20:
@mossmann Depending on how soon the next release comes, I would propose a bootstrap/bootloader that configures the CPLD from a bitstream in SPI flash every time, ensuring the firmware and CPLD match. I have some larger plans for said bootloader, but, one thing at a time...

mossmann, 2018-02-20:
@jboone We're definitely thinking about that, but we don't want it to hold up the next release.

schneider42, 2018-02-20:
Won't that cause lot's of wear on the flash cells? Is there a way to first verify the bitstream?

jboone, 2018-02-20:
@schneider42 There's a provision in every flash-based CPLD I've used (CoolRunnner-II included) to load a bitstream directly into the RAM cells that configure the device. This would not entail reflashing the CPLD. This technique is already employed in the PortaPack firmware, FWIW. Look for a Xilinx document called "CoolRunner-II Programmer Qualification Specification", which discusses SRAM read and write operations.

mossmann, 2018-03-28:
added to release notes for next release

commit fc8fcd86 (2018-02-20): added CPLD update requirement to firmware/README
improved cpld/README
fixes #458
closed 2018-03-28 by mossmann