greatscottgadgets/hackrf #459: recent hackrf_exit() returns "some hackrf is sill in use"

guruofquality, 2018-02-21:
Looks like hackrf_exit() never decrements open_devices. Its causing some strange errors for me, even when hackrf is not even used because I print the error code:
* reference https://github.com/pothosware/SoapyHackRF/issues/20

Added by commit: https://github.com/mossmann/hackrf/commit/73ca8c09 perhaps?

guruofquality, 2018-02-27:
@gvz So hackrf_init increments the call, and hackrf_close decrements the call. But these arent matching calls, so naturally hackrf_exit() will return an error.

example: finding devices
```
call hackrf_init()
call hackrf_device_list_open()
hackrf_exit() - gives an error, open_devices never decremented
```

example: opening two hackrf
```
call hackrf_init()
d1 = hackrf_open()
d1 = hackrf_open()
hackrf_close(d1)
hackrf_close(d2)
hackrf_exit() - gives an error, open_devices is negative
```

Judging from the variable name, `open_devices++` should be in the actual open* calls and not hackrf_init? That way it would count the actual number of open devices.

dominicgs, 2018-02-27:
I agree, it looks like moving the `open_devices++;` line to the end of `hackrf_open_setup()` should do the trick.

gvz, 2018-02-28:
Oh, I guess that was my fault.
I used gnuradio to test my code, it calls init and open equally often.
At the moment I can not perform any tests, as all my hackRFs a occupied in measurements.
Therefore, I hesitate to make a pr that fixes this issue.

dominicgs, 2018-04-01:
I believe that I  fixed this in 65c7b5a

closed 2018-04-01 by dominicgs