greatscottgadgets/hackrf #463: Incorrect behavior stop_rx() and stop_tx()

cotara, 2018-03-14:
### Steps to reproduce
1. call hackrf_open_by_serial(serial1, devise1);
2.call functions such as hackrf_set_sample_rate(), hackrf_set_freq(), etc...
3. call hackrf_start_rx(device1, rx_callback, NULL); The device1 was successfully started at the reception.Orange LED on the device1 is turnd on
4. Repeat steps 1-3 for device2. The device2 was successfully started at the reception.Orange LED on the device2 is turnd on. Both devices visit to rx_callback(transfer) with different transfer. Everything is work.
5. call stop_rx(device1). Then device1 stop correctly. Orange LED on the device1 is turned off. Orange LED on the device2 is turned on. But both devices now do not attend rx_callback(transfer)!!

### Expected behaviour
only device1 should be stopped!

### Actual behaviour
both devices now don`t visit rx_callback(transfer), though orange LED is turned off only in device1.

### Version information
**Operating system**: Windows10

**hackrf_info output:**
Found HackRF
Index: 0
Serial number: 0000000000000000a06063c8224eb75f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00684764

Found HackRF
Index: 1
Serial number: 0000000000000000a06063c8254a615f
Board ID Number: 2 (HackRF One)
Firmware Version: 2018.01.1 (API:1.02)
Part ID Number: 0xa000cb3c 0x00514367

If you are reporting a problem that involves third party software 
(GNU Radio, Gqrx, etc), please report the version here.

We write our program with transfer.c as an example.
Should we use separate callbacks for each Hack RF? 
It seems to us that the problem is in threads libusb. When we stop one device, the threads of other devices are corrupted.

dominicgs, 2018-03-14:
It looks like we set the `do_exit` flag when we stop a thread and this causes the other thread to also exit.

I think it should be a value on the `hackrf_device` struct so that we avoid this.

cotara, 2018-03-15:
This is the first thing we did!
We've added an option do_exit to hackrf_device struct and now we can change it for each device.
Also we have changed func request_exit(hackrf_device* device):
static void request_exit(hackrf_device* device){
    device->do_exit = true;
}
And now we have strange thing: 
1.We start device1, then start deivce2, then stop device2, all is good, the again start device2, is good, stop device2, is good.... and after 3-10 times problem appears.
2. We start device1, then start deivce2, then stop device1 (that is, other than we ran last) and the problem appears at once!

Also we try change hackrf_libusb_transfer_callback in this part:
else {
		/* Other cases LIBUSB_TRANSFER_NO_DEVICE
		LIBUSB_TRANSFER_ERROR, LIBUSB_TRANSFER_TIMED_OUT
		LIBUSB_TRANSFER_STALL,	LIBUSB_TRANSFER_OVERFLOW
		LIBUSB_TRANSFER_CANCELLED ...
		*/
Also we try reboot the device after every stop. No effect...no more ideas.

dominicgs, 2018-03-21:
Do you have a simple example program that triggers this behaviour?  I'm struggling to reproduce this bug.

closed 2019-12-10 by miek