greatscottgadgets/hackrf #466: Which chip does switchable image reject filter use?

FriedMushroom, 2018-03-19:
When i study the circuit of hackrfone, i can't find which chip does switchable image reject filter use? Or please tell me how does switchable image reject filter works?

H1Rong, 2018-03-19:
I'm also confusing. How could the LP and HP adapt the frequency we want? The chip I found someone said for HP is DEA162400HT-8004BA and for LP is LP0603A1880ANTR. But I found the frequency for these two are 2400MHz- 2500MHz,1840MHz-1920MHz respectively. 
So what about the frequency does not in these range?
Could I get the right chips we used for LP and HP? Because I'm focusing on the study of getting the BER or Eb/N0 for the receiver and urgent to know the character of the filters.
Thanks.

mossmann, 2018-04-05:
The high-pass filter is DEA162400HT-8004B1.  The low-pass filter is LP0603A1880ANTR.  They are switched with a pair of SKY13350 RF switches.  This information is contained in the KiCad source files.

For information about the frequency response of these filters, refer to their datasheets.

closed 2018-04-05 by mossmann