greatscottgadgets/hackrf #472: Finding and removing 25-50k worth of buffers in the tx/rx chain

cedivad, 2018-04-09:
I'm using `hackrf_transfer` and changed `lib_device->transfer_count`/ `lib_device->buffer_size` in `hackrf.c` to basically zero. My host application should have close to zero latency. 

What else should I look into to decrease the tx/rx buffers further? I'm using the device in 4 MHz mode so I assume I have some overhead with respect to the defaults as those are probably optimised to work for 20 MHz. 

I've surfed the firmware code very briefly and found this:

https://github.com/mossmann/hackrf/blob/5fbd75ef94b9815af19a839aca21e3679c9b5151/firmware/hackrf_usb/hackrf_usb.c#L223

The maximum amount of data that call will transfer set to ~16k. Could that be responsible for the buffers I'm seeing? Should I look anywhere else?

I'm sorry for asking such silly questions, I plan on looking into it more tomorrow and will update this if I find anything useful.

dominicgs, 2018-04-09:
Yes, the buffer size in both the host code and HackRF firmware will need to be changed.  There are some assumptions we make about those buffer sizes, [here](https://github.com/mossmann/hackrf/blob/master/firmware/hackrf_usb/usb_bulk_buffer.c#L250 and [here](https://github.com/mossmann/hackrf/blob/master/firmware/hackrf_usb/usb_bulk_buffer.h#L32)

cedivad, 2018-04-09:
Thank you Dominic, that seems super promising. 

However I have an unrelated problem (that I understand is off topic but I don't think I should start a new thread), as I've tried following the firmware build instructions without any code changes... And they work... Half of the time. 

I'm somehow dropping frames with the firmware I built myself when receiving. I've tried this a couple of times with a newly cloned repository (again, no code changes) without any luck. Installing the binary provided fixes everything, but mine doesn't want to work.

I'm using osx, maybe that's the problem? Is there a suggested build environment? Ubuntu 16.04? (I've began downloading that).

cedivad, 2018-04-10:
Ok, turns out I was using master for my firmware and the last release for my host application. Solved. 

I decreased those buffers to basically zero as well (0x8000 > 0x1000), but my latency only decreased by ~10%. Are there other buffers I should look into other than the firmware ones listed above and the ones defined in the host's hackrf.c? Or is my latency coming from my application itself?

Thank you.

dominicgs, 2018-04-10:
Does this mean all of your questions have been answered?

Are you happy if I close this issue?

closed 2018-04-10 by cedivad